Jyers/Marlin #254: How are different releases built?

bjoernbusch, 2021-03-14:
I love the different FW builds that you are producing and providing! Thanks for that!

Any chance you can elaborate a little bit, how you produce the different kinds of releases (with probe, no probe, different amounts of probing points, etc.) from the stored sources? In the configuration folder I only see one set of configuration files from you.

Thanks!

t3chguy, 2021-03-14:
Stuff like this should go in "Discussions" - not "Issues".

The configs given have a configuration matrix applied on top of them to generate all the permutations - https://github.com/Jyers/Marlin/blob/Ender_3_V2_Extensible_UI/.github/workflows/compile-configs.yml#L69

bjoernbusch, 2021-03-14:
Sorry, didn't see the discussions tab. 

Thanks!

closed 2021-03-14 by bjoernbusch

Jyers, 2021-03-15:
In case you're interested, I don't think techguy's answer fully answered your question. The folder labeled Jyers's Configs has my personal config files, the config files for all the different builds can be found in the E3V2 Templates folder. This contains the primary skews, while the small changes like grid size and board version are handled by my build matrix.

bjoernbusch, 2021-03-15:
Yes, after following the T3chguy's pointer, I could find all the info I was looking for. I didn't realize that this repo is so well structured with Discussions and Wiki and Actions and so on. Usually I only see repos with plain sourcecode and certain magic isn't even checked in.

I truly appreciate this repo, I could finally shed some light on my leveling issues with the 15x15 UBL build. With the standard BLTouch, I could always only see an issue on the right, now I see that I have issues all over the bed and it looks like an issue with wobbling rollers or so. So now I can analyze it much better. Thanks!