Jyers/Marlin #259: Stuck probing off bed?

callumj, 2021-03-14:
### Description

Hardware: Ender 3 V2 with `E3V2-UBL-BLTouch-10x10-v4.2.2` release (tried latest and previous UBL release).

When doing a level (create new mesh) with UBL enabled it eventually gets to a point where it goes off the bed and just continues to probe in that same point (point X/100 keeps increasing but it never moves along the axis) as if it is stuck.

[image]

### Steps to Reproduce

1. Go to Level
2. Enable UBL
3. Create New Mesh
4. Eventually it gets stuck on probing some area off bed and continues to probe

#### Additional Information

Using release firmware

callumj, 2021-03-14:
Issue with `MESH_INSET = 0` in `Configuration.h` - compiled it myself with `MESH_INSET = 10` and the probe isn't going off the bed.

closed 2021-03-14 by callumj

Jyers, 2021-03-15:
Just an FYI, this is most likely due to incorrectly set nozzle to probe offsets rather than the mesh inset setting. The offsets can be set both in the config files and in the Control>Advanced menu on the display.

robhuska, 2021-04-13:
> Just an FYI, this is most likely due to incorrectly set nozzle to probe offsets rather than the mesh inset setting. The offsets can be set both in the config files and in the Control>Advanced menu on the display. This is an important distinction, as incorrectly set nozzle to probe offsets will result in a shifted mesh that will not align with your print bed.

As a note for this, I had to update my X Offset to 43mm. Now the probing seems very close to exact center of my bed.