Jyers/Marlin #264: Manual Mesh Leveling slower than a snail pace & off back of bed

Legsmaniac, 2021-03-15:
### Description

I finally got around to trying the latest Extensible UI.  I have self-compiled using the E3V2 UBL-NoProbe-3x3 template as a starting point, making only minor changes.  
When doing Manual Leveling from the Prepare menu, the speeds are fine.
However, when doing a UBL Mesh leveling, the speed is REALLY slow.

### Steps to Reproduce

1. Select Level from the main menu
2. Go through the process
3. See just how slow the head moves.   I could make a cuppa tea in between leveling points.  That's 8 cups o tea per meshed bed!

**Expected behavior:** [What you expect to happen]

I'd expect it to be faster than what it is!  At least as equal to the normal Manual Leveling which isn't all that fast but WAY faster than Mesh Leveling. 

**Actual behavior:** [What actually happens]

As described above - REALLY slow.
Also, when the last 3 points are reached (back of the bed) the nozzle is off the bed by about 3 or 4 mm.
I have also mentioned before in another issue that the nozzle is too close to both front and back that it hits the bed clips.
Could this be something to do with the bed size being set to 230x230 as default when it should really be 220x220?

#### Additional Information

Personally I would also have prefered Mesh Level points on the bed to be the same as Manual Leveling, which is where the nozzle is placed directly over the screws.  Is this something that could be done and would it be a better idea?

Turismo05, 2021-03-15:
Mines is the same, it placing the nozzle right at the corners of the bed. would be much better over the screws as you say.

feelfree69, 2021-03-15:
Same here. Unbelievable slow, close to unusable.
And the measurement points should be in the middle of a measurement square, not at the outest possible position.
For example, with a 5x5 bed, the measurement postitions should be at 10, 30, 50, 70 and 90% of an axis.
I tried only the 3x3, here the postions are 0, 50, 100 instead of 25, 50, 75. 
0% and 100% are likely to be outside the bed.

cobyn, 2021-03-15:
same here, it was fine when i tested it after flashing, but the next day it was very slow

Jyers, 2021-03-16:
This will most likely just boil down to some config changes for the ubl templates. Thanks for the reports, I'll do some testing and see if I can get most of this fixed.

Jyers, 2021-03-17:
The latest source should contain the fixes for all the issues. I increased the movement speed and added a margin around the bed. However, be aware that if your probing grid was going off of the bed that is a clear sign that your probe offsets are not set correctly, I would advise you to make sure you get those correct otherwise your mesh will be offset in relation to the actual print surface.

closed 2021-03-17 by Jyers

Legsmaniac, 2021-03-17:
Thank you for the fix. Will try it out in the morning.

Probe offsets? Not sure I understand this. There is no probe. Well, the nozzle is the "probe" in the case of manual mesh, isn't it? In which case, wouldn't all offsets be zero? 

When I homed (G28) the nozzle was right on the left bottom corner of the bed (0, 0) so I would expect the nozzle to be on the back of the bed rather than a few mm off it.

Jyers, 2021-03-17:
Guess I missed the part of your description where you mentioned using the UBL-Noprobe... 
That's very strange though, unless your homing point is super off, or your bed is less than 230x230 (the stock bed it 235x235). I can't imagine why that would be happening.

Turismo05, 2021-03-17:
Mines made the bed (just) at three of the corners but missed on the top right.

Legsmaniac, 2021-03-17:
> I can't imagine why that would be happening.

Well it seems I'm not the only one.
OK, flashed new firmware, speed is MUCH better, thank you. 
But I still have the problems where the nozzle is right on the edge of the bed/off the edge of the bed/over the clips/hitting the clips.

So I carefully checked my home point - it's a few mm into the bed and not exactly on the corner, which would explain why it's off the bed at the other end.  Seems the Y Axis end stop cannot be adjusted, it's fixed in place.  If I could move the end stop back a few mm, it would solve my (and others) problem of point accuracy but not the clips problem.  I imagine there must be various bed tolerances during production which is why some people say it's right on the end and some say it's off the edge. I still say it would be better inside the bed confines a bit more, preferably over the screws like others have said.

Is this possible please?  Is this something I can alter myself in the config files somewhere?
Thanks again.

feelfree69, 2021-03-17:
My bed is 235x235, but the official active print area is 220x220. It does not make any sense to put the sensor outside of this 220x220 area. 
The position of the screws seems to be a reasonable good measurement point.

Jyers, 2021-03-17:
If you are using your own config, you might not have gotten the fix, you need to set this "#define MESH_INSET 20" in configuration.h This defines how far in the mesh should be from the edges of the bed.

Jyers, 2021-03-17:
Also for your issue with the endstops, considering you are compiling your own, my solution is to set the min axis positions to match the actual positions on the bed of the endstops, here's my config:

#define X_BED_SIZE 231
#define Y_BED_SIZE 228

#define X_MIN_POS 4
#define Y_MIN_POS 7
#define Z_MIN_POS 0
#define X_MAX_POS (X_BED_SIZE + X_MIN_POS)
#define Y_MAX_POS (Y_BED_SIZE + Y_MIN_POS)
#define Z_MAX_POS 250

Note that this will cause any movements at the far left or front of your bed to be cut off (ie purge line)

feelfree69, 2021-03-17:
For the endstops: I think there's a valid reason why max. print area is 220x220 by default. Anyone who want to use more than that could do it by compiling his own firmware, but the default should always be a safe and easy one, which erverybody can handle without tweaking.

Legsmaniac, 2021-03-17:
> If you are using your own config, you might not have gotten the fix, you need to set this "#define MESH_INSET 20"

I did indeed comfile my own and seems I missed that setting!  My bad!  I used WinMerge to compare your Configuration.h in  UBL No Probe 3x3 in Configurations folder with my own.  I must have accidentally skipped past that one.  I set #define MESH_INSET 27 which places the nozzle roughly over the screws which I prefer.  Although now I think about it, this is not a bed leveling procedure but a bed mesh setup so it's probably better wider, 15 should probably be fine since that will avoid the clips.  Saying that, most of my prints are usually within the confines of the centre of the bed. I've yet to print anything that reaches the edges but there's always a first time.......

> Also for your issue with the endstops, ..... here's my config:

Excellent, man!  I could probably fine tune that to my own bed/printer but for now, I just copied your suggestion and it's pretty good as it is really.  I have to say, my prints have been better since using your firmware and more so since using UBL Manual Mesh.  It's way better than BLT IMHO.

Thank you once again, for the fixes and suggestions.  Very happy.  
Could this firmware possibly get even better yet?  It's hard to imagine.  👍😃

Jyers, 2021-03-17:
> For the endstops: I think there's a valid reason why max. print area is 220x220 by default. Anyone who want to use more than that could do it by compiling his own firmware, but the default should always be a safe and easy one, which erverybody can handle without tweaking.

I see your point, but 220x220 is excessively small. Hence why I have it set to 230x230 by default, still safe, but gives you a larger print area if you need it. Plus if you really want to be safe, you can just make it 220 in your slicer.