Jyers/Marlin #268: Can't enable both MEATPACK and BINARY_FILE_TRANSFER

jjfs127, 2021-03-16:
Enabling both MEATPACK and BINARY_FILE_TRANSFER generates the followng:

```
In file included from Marlin\src\HAL\STM32F1\../../inc/MarlinConfig.h:49:0,
                 from Marlin\src\HAL\STM32F1\HAL.cpp:30:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:3311:4: error: #error "Either enable MEATPACK or enable BINARY_FILE_TRANSFER."
   #error "Either enable MEATPACK or enable BINARY_FILE_TRANSFER."
```

Likely a bug with Marlin but I'd like to register it here.
Thanks

t3chguy, 2021-03-16:
That's not a bug with Marlin but a known incompatibility. Enabling both would break serial communication hence the `SanityCheck` warning preventing you from doing so.

jjfs127, 2021-03-16:
I guess curious as to why it was enabled in Jyer's most recent commit: https://github.com/Jyers/Marlin/commit/c5bca319c2fcffe223db58ffbb3b26e3080e79c3

t3chguy, 2021-03-16:
That commit disables BFT and enables Meatpack. It does *not* enable both.

jjfs127, 2021-03-16:
> That commit disables BFT and enables Meatpack. It does _not_ enable both.

Yeah, you're right. My bad.

jjfs127, 2021-03-16:
Just hopeful we'll be able to get meatpack and BFT working together to allow us to upload firmware files....

Jyers, 2021-03-16:
I never managed to get firmware updates working from a host device, even with BFT, so I decided to swap it for Meatpack support.

closed 2021-03-16 by Jyers

Sennevds, 2021-03-17:
By the way @Jyers the latest version of the firmware updater plugin works for the ender 3 v2 have been using it for 2 weeks now

Jyers, 2021-03-17:
@Sennevds Really!?!?!? I have tried it so many time and never gotten it to work... Does it need binary file transfer? Also what settings are you using? Sorry for all the questions, but if I could get that working it would make my life so much easier!

Sennevds, 2021-03-17:
Yeah you need to enable binary file transfer (not compatible with meatpack) you need to manually install the python packages(https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/blob/master/doc/marlinbft.md#installation) here are my settings for the plugin it self:
[image]
[image]