Jyers/Marlin #271: Knob button randomly engaging itself

vhristenco, 2021-03-16:
I'm running 1.2.1 with Jyers's config on a 4.2.2 mainboard and sometimes, during the print, the printer beeps and enters the Tune menu by itself.
Just finished a 3h print and this happened 4 times.

Schnabulation, 2021-03-16:
I'm sorry but I guess this is a hardware issue: I flashed the latest version yesterday and did a 7 hour print - no issues at all.

Majestry, 2021-03-16:
I noticed this problem once from about 50 hours of printing. Also during the print, the beep sound appeared and tune menu was opened.

Jyers, 2021-03-16:
As @Schnabulation mentioned, this is almost certainly a hardware issue. I recommend checking your encoder for any signs of inconsistency or flakey behavior and also checking you connection between the LCD and the mainboard, as a bad connection or running the LCD cable too close to the heater wires could also cause this issue.

closed 2021-03-16 by Jyers

vhristenco, 2021-03-18:
However, the printer never returns to the main screen after entering the "tune" menu (which is supposed to happen if it would be an encoder issue, as the first menu option is "back"). Also, doesn't happen on other firmwares.

Jyers, 2021-03-18:
From your original report, you said this happened 4 times. I'm assuming this means you put it back to the print screen after each occurrence. If this issue persists, try not putting it back to the print screen and see if it does in fact do it on its own.
Also considering the the code for the encoder is nearly identical to the stock marlin code, I find it hard to believe this is unique to this firmware. If the issue does persist, try flashing back to a different firmware and see if the issue still exists. If it does not then I can have you try a version without the small change to the rotary code and see if that helps.

vhristenco, 2021-03-19:
I just tried the latest Marlin and the Smith3D firmware, and the issue doesn't occur in either one of them. 
FYI, I compiled your firmware myself, using the config files from the "Jyers Config" folder and applying a few tweaks (steps/mm, M600 config and such). Printing from SD card or Octoprint makes no difference, the printer enters the "tune" menu by itself after a while. If I just let it there, it won't return to the main screen. If I click on "back" and return to the main screen, it will go again to "tune" menu after a while.

Jyers, 2021-03-21:
Do you find that this issue occurs with all gcode or just specific files? My apologies for all the questions, but as very few people seem to be experiencing this issue, I'm trying to pinpoint what may be different about your config.

vhristenco, 2021-03-22:
This is totally random. The same file could print without issues or make the printer enter the menu 2-4 times.

vhristenco, 2021-03-23:
Update: It also exits the tune menu by itself (just happened).

Jyers, 2021-03-23:
The fact that this seems to be occurring entirely at random still points me to believe this is a hardware issue of some kind. As nothing to do with the firmware, especially something as cut and dry as an encoder click, should result in random triggers. And as there have only been two reports of this, I can only imagine its something specific with your setup. I would still very much like to help fix this, but as far as I can tell, there is nothing in the firmware that could produce random occurrences like that.

vhristenco, 2021-03-24:
Just switched to Smith3D's firmware last night. No problem so far (printed about 4 hours without the issue to occur).
But with that firmware I get random bltouch errors (randomly doesn't engage when probing, ending in the nozzle scratching the bed surface). Your firmware solved that issue, but makes random clicks.... Seems that I have to choose between the two.

Schnabulation, 2021-03-24:
...or: you could learn how to compile your own firmware and take the best of both worlds. Wouldn't be too difficult to change the rotary knobs binaries from the upstream Marlin and the rest from this firmware.

Jyers, 2021-03-24:
[BLTouch-3x3-HS-EncoderTest.zip](https://github.com/Jyers/Marlin/files/6197807/BLTouch-3x3-HS-EncoderTest.zip)

Give this a try. It'll break some things, but the general stuff should be fine. It'll let me know if the issue is the encoder file itself.

vhristenco, 2021-03-25:
I just replaced the two rotary_encoder files with the ones from Smith3D. I'll post back tomorrow... let's see if anything changes.

Jyers, 2021-03-25:
You most likely put them in the wrong place. The dwin code is now located in Marlin\src\lcd\extui\lib\creality_dwin Putting them here will require some include changes, hence why I uploaded a bin for you to test.

vhristenco, 2021-03-26:
Well, that seemed to solve my issues. Printed about 10 hours without entering the "tune" menu :)
Looking at the code, I'd say you didn't change much, but.... maybe you should revert to the old code.
Thanks again for your great firmware!

Jyers, 2021-03-26:
Although this is good to hear, I'd keep printing with that for a bit to make sure the problem doesn't come back. I would still like to believe its not the code. So if you don't mind one more test I saw someone mention this in a discussion on this issue

>Hi I had this problem with Creality firmware, it beeped and went to tune and back constantly during prints. I realised it was the press switch on the encoder being very sensitive. I turned the printer off and pressed the knob firmly loads of times and this cleared the fault! I think there may be debris inside the mechanism randomly triggering the button.

Give this a try with the original version once you have thoroughly tested the modified one. Thanks for all your help with this, if it does end up being the file I'll figure out the best way to revert it, I'm just trying to avoid that because it would break some popup interactions in the firmware.