Jyers/Marlin #275: UBL for dummies

Notyly, 2021-03-16:
Hello,

I am now a proud user of your firmware.

Now I have two questions for you:

1:
What script should I use, if want to delete the old and probe a complete new mesh and store it, while using bed visualizer on octoprint?

2:
What start script in Cura should I use to only get the tilt adjustment (2x2 or 3x3) before the next print?
(You suggest M420 S1 in your Wiki, but I always read G29 J as suggestions)

Thanks a lot.
Patrick

Jyers, 2021-03-17:
Give this a read if you haven't already: https://marlinfw.org/docs/gcode/G029-ubl.html
That should give you a rundown of what commands you need to do what. (The examples are an especially good place to look)
I believe G29 P1 followed by a few G29 P3 will be effective for a Bed Visualizer script
G29 J will autotilt the mesh, so if you want to autotilt before every print that is what you want. Otherwise M420 S1 will just make sure the mesh is enabled.

waywit, 2021-03-17:
Hi, 

I have started with this reddit post, especially the second part of it, to understand how UBL works: https://www.reddit.com/r/ender3/comments/i9xt6e/stepbystep_guide_for_ender_3_v2_firmware_w/

The first part is about compiling the firmware on your own (which you do not probably need), but the second part describes how to initially setup UBL on your printer and then there is an example start script for Cura (loading the mesh and revalidating 3 Points) to insert before every print... I have transferred it to my own to PrusaSlicer script... The comments helped me a lot to understand the G-Code...

closed 2021-03-24 by Jyers