Jyers/Marlin #295: Auto leveling doesn't work after manual mesh.

nicola-p24, 2021-03-19:
I installed the .bin "manual mesh" 4.2.2 but it doesn't work.
It runs the mesh configuratoor correctly, I save the settings and put M420 S1 command after G28 in the g-code but it doesn't work.

The z-axis does not compensate during the printing.
It doesn't work with both 3x3 and 5x5 versions.

tititopher68, 2021-03-19:
Hy ;-)

I think you have some issues with your own Configuration.h and/or Configuration_adv.h
Look at these 2 files : https://github.com/Jyers/Marlin/tree/Ender_3_V2_Extensible_UI/Configuration%20Files/E3V2%20Templates/ManualMesh-3x3
With these files, no issue. M420 S1 work well
(Sorry for my english but i am french)

Jyers, 2021-03-19:
@tititopher68 I believe they are using a precompiled version.

As far as the issue goes, are you sure the z is not compensating? It can be quite hard to tell if you don't know where to look. Try a test print that uses most of the bed and watch the z-axis coupler to see if it turns. It will be very slight, but it should move.

nicola-p24, 2021-03-19:
> @ tititopher68 Credo che stiano usando una versione precompilata.
> 
> Per quanto riguarda il problema, sei sicuro che la z non stia compensando? Può essere abbastanza difficile dire se non sai dove guardare. Prova una stampa di prova che utilizzi la maggior parte del letto e osserva l'accoppiatore dell'asse Z per vedere se gira. Sarà molto leggero, ma dovrebbe muoversi.

Yes, i'm using a precompiled version. This evening i'll try to delete "RESTORE_LEVELING_AFTER_G28".

I'm sure that z axis doesn't move and the printing confiirming this.

Jyers, 2021-03-19:
What do you mean by g29 doesn't work?

nicola-p24, 2021-03-19:
To view the mesh calibration

Jyers, 2021-03-19:
To view mesh calibration use M420 V1

nicola-p24, 2021-03-19:
Ok, I'll check it out later.
Thanks!

nicola-p24, 2021-03-19:
To recap: I would like to do "manual bed leveling". I download the source code and replace configuration.h and configuration_adv.h with the files I find at this link: https://github.com/Jyers/Marlin/tree/Ender_3_V2_Extensible_UI/Configuration%20Files/E3V2%20Templates/ManualMesh-3x3.

If I wanted to make a calibration to 5x5 I replace the value in the line #define GRID_MAX_POINTS_X "5".

Correct?

Jyers, 2021-03-19:
Yes, that is correct

tititopher68, 2021-03-19:
Just a precision, in your StartGcode, you must use M420 S1 not G29 after G28 !

GeeJeeDee, 2021-03-19:
M501 before not necessary?

nicola-p24, 2021-03-19:
[image]

If I write M420 V1 I don't see anything. I would see the grid like in photo, how can I do?

In my opinion z axis doesn't move yet.

tititopher68, 2021-03-20:
Hy Nicola,

Config files looks OK
Are you sure you have a v4.2.2 motherboard ?

I give you this link.
I compiled a firwmware which integrates a Mesh Viewer (Menu Control -> Advanced), the rest is identical to Jacob's.

https://e.pcloud.link/publink/show?code=XZG4XXZjDiCddgnMz514FFQrOLSSFHjBj9y

First you need to flash your LCD screen : 
Switch off the printer, disconnect the cable from the LCD screen, 
Unscrew the 4 screws, unclip the box, 
At the top right is the MicroSD port. 
Format a MicroSD card (8 or 16 GB) in FAT32 - 4096 Bytes,
Copy to the MicroSD the DWIN_SET folder ("Screen Firmware" folder) and the E3V2-Manual_Mesh-3x3-v422.bin file,
Insert the MicroSD card, 
Reconnect the screen cable, 
Turn on the printer,
The screen will appear Blue then a few seconds later Orange.
Turn off the printer,
Remove the MicroSD

Now flash your motherboard : 
Insert the MicroSD card in the Port on printer,
Turn on the printer.

First, go to Control and do : Reset Settings, then Store Settings, then Restore Settings.
Go to Prepare, Prehead PLA. 
Do Autohome, then Manual Leveling and level the 4 corners.
Then go to Z-Offset and Activate live adjustment, then adjust your Z-Offset with Microstep Up and Down, then Save.
Go to the main Menu, then Level and do your Mesh Bed Leveling.
Go to Control and Store Settings.
Go to Control and View Mesh

In your StartGcode, after G28; place M420 S1 and test a print.
Depending on the flatness of the bed, you should see the Z axis motor moving (micro-rotation in one direction and / or another).

But if you want to make it more visible, redo a Manual Leveling in Prepare and leave a little more play between the nozzle and the bed (with your knobs).
Redo a Mesh Bed Leveling (without forgetting to save) and redo a printing test. You should see more of the Z axis being moved.

Good luck !

nicola-p24, 2021-03-20:
@tititopher68 thank you for the amazing tutorial.
Yes, I have checked the version of the motherboard and it is 4.2.2. The only one thing that I never done is the flash of the LCD screen.
I followed a tutorial on YouTube in which the person was updating his ender and the procedure is this:
-Switch off the 3d printer;
-Download the .bin file;
-Formatting SD card;
-Copy .bin into SD card;
-Insert SD into micro usb slot
-Switch on the 3D printer and the firmware i'ts updating after 1 minute.
I've do in this way and the firmware was correctly updated.
*I have the stock motherboard

nicola-p24, 2021-03-20:
Thank you @tititopher68 and @Jyers 
Everything is working perfectly now!!! The Z axis now turns and compensates
[image]

closed 2021-03-20 by nicola-p24

Jyers, 2021-03-21:
Happy to help and glad you got it working. Happy leveling! 😄