Jyers/Marlin #298: FR: Z offset for Manual Leveling?

DerPit, 2021-03-20:
Disclaimer: I'm new to this, so I hope I'm not missunderstanding too many things :p

Having a BLTouch, I assume the 'Manual Leveling' is to get the bed as level as possible to start with.  So I use a probe (paper etc.) and adjust the knobs to get a constant distance between nozzle and bed everywhere.
I see the pre-defined positions always go to Z=0.  Wouldn't it be logical to set that to Z=thickness of my probe, and make that value adjustable in that menu?  At least that's how I had done leveling before the BLT:  Home Z, then go to (X,Y,0.1) and adjust...

tititopher68, 2021-03-20:
First preheat your bed and head.
Go to Prepare -> Manuel leveling ans adjust the 4 corners with the scroll wheel and a paper.
Go to Z-Offset and adjust your Z-offset (live adjustment enabled) with Microstep Up and Microstep Down. Don't forget to Save.
Go to Main Menu and do a Level.
At the end, go to Control -> Store Settings.
In your StatrtGcode in tour Slicer, After G28, place a M420 S1 or M420 S1 Z2

Jyers, 2021-03-21:
While this does seem logical, as far as I'm aware, for print jobs the print uses z=0 for the first layer, even if it may say 0.2 or whatnot, I don't think it actually moves up. This comes from the assumption that if you send the printer to Z=0, it is not and should not be scraping on the bed, which would be the case if it actually used 0 as bed height. Either way, any slight discrepancy in the offset when leveling your bed should be easily accounted for using your z offset. Manual leveling is really more for getting the bed flat rather than getting the nozzle offset perfect.

DerPit, 2021-03-21:
Hmm, no - the way I did it (level with Z=0.1) I never had issues with the nozzle hitting/scratching the bed.  And the first 'real' command (after the init line) is, e.g., "G1 Z0.200 F9000.000" for a print with first layer height 0.2. At least my slicer (prusa-slicer) generates it like that.

My point is that if the manual leveling is done with Z=0, the plate will go down 0.1 (or whatever the probe thickness is), as going to that menu will always first do a Z homing.  So you are at 0, then move the bed down 0.1.  Do it 10 times, and your bed is 1mm lower than at the beginning.  At some point you'll be at the limit :-/

Yes, I'm aware I could change the Z offset, and set it back afterwards.  Just, doing that manual leveling at Z=probe_height would seem much more logical to me....

(I agree that is different if the real external Z endstop is used, as that has an absolute, fixed position.  But the BLTouch only does a relative calibration between bed and probe...)

Jyers, 2021-03-24:
I've been thinking over this for the last few days and I think I've finally come to the reason this doesn't fully make sense to me. I get the concept, but I think the reason this feels unnecessary is this: The best procedure for manually leveling your bed is generally to tune in your z offset using the z offset tool, and then manually level the bed. Here's where I see the redundancy. When you tune the z offset, you do so using a probe of some kind, lets just say its a piece of paper of 0.1 mm thickness. What you are essentially doing when setting your z offset using that probe is setting the zero position of your printer to 0.1 mm from the bed. Then once you start manually leveling, the nozzle will already be one papers width off the bed and you can simply level the bed as you would, the bed should not move up as a whole. 

This is just the conclusion I've come to, but if you still believe an offset for the menu would be beneficial I would love to hear your thoughts on this.

DerPit, 2021-03-24:
Thanks for thinking about this.  I agree, it has a lot to do with personal habits of doing things.

I'm in particular talking/thinking about the BLTouch version, where you don't do a manual ("paper-based") Z offset tuning.  Instead, the automatic homing is triggered when you enter the 'Manual Level' unit (unless it had already been done before).  The height offset between nozzle and sensor is fixed, and well-calibrated.  So after that, zero is zero, i.e., going to z=0, the nozzle will just touch the bed.  So I first have to go to the Z offset menu and change the offset there (trying to remember whether I have to change -3.06 to -3.16 or -2.96).  Then go to the leveling menu, do the leveling, and after that go back to the z offset menu to set it back to the old value.  That's why I think having a dedicated "leveling height" (that is only used in that menu, so Z offset is not touched at all) instead of a hard-coded "Z0" would be clearer for the user, and save some additional steps.

Before I had the BLTouch, I would first zero the axes (which is an absolute reference, also in Z), then set Z=0.1, do a 4-point leveling at the XY corners, and be done.  I've never actually used Z offset, it was always zero (just like now it always is just the sensor-nozzle offset).

So I think for people that work like I do, such an offset for manual leveling would be a great addition, and (defaulting to offset=0) it wouldn't change anything for others.  I would absolutely use it, but if my routine is odd, and/or based on some miss-conception, and I'm the only one doing so it's surely not worth the effort.

Jyers, 2021-03-26:
Interesting, I think the discrepancy between our methods here lies entirely in how we see the Z Offset. For as long as I have been printing I have always found that having the z offset about a papers width off the bed has produced the best prints. If I were to put it all the way down to just touching the bed my nozzle would be far too low and my first layer would get wrecked. This is with a BL Touch by the way. I also wouldn't describe the Z offset as fixed. Different filaments require different nozzle gaps to produce a good first layer. For instance petg requires a decently larger gap. This is generally handled by adjusting the Z Offset slightly. This being said, I think manual leveling with paper when a BL Touch is available is outdated anyway. I would like to add a G35 tramming assist like functionality to the display which should make paper leveling unnecessary.

samsp-msft, 2021-03-27:
> I would like to add a G35 tramming assist like functionality to the display which should make paper leveling unnecessary.
👍

DerPit, 2021-03-27:
Hehe, as I mentioned - I'm quite new to 3D printing.  So obviously I'm missing some experience that you have.
I have however so far not had issues with wrecked first layers (unless they really had too little surface and required a brim).   I haven't used filament other than PLA+/Pla Carbon so far, though.
But coming back to the original request:  Either way it seems one would have to use some Z offset for the manual leveling, whether it is changing the Z offset or the Z value of the mesh points.  So IMO it would be straight forward to have access to that from within the menu.

G35 support via the BLTouch would also be a cool thing for sure.  One grain of salt there of course is that the screw positions on the far X side aren't reachable with the probe :(

Jyers, 2021-03-27:
Alright, you've convinced me. I added a Z position setting to the manual level menu

closed 2021-03-27 by Jyers

DerPit, 2021-03-31:
Just as Feedback:
Thank you so much for this!  I've finally managed to compile, and then dared to install what I compiled :D 
I didn't brick my Ender, and the manual meshing - for my use - now really is perfect.