Jyers/Marlin #299: Backing out of Manual Leveling Produce M112

jjfs127, 2021-03-20:
After upgrading to v1.2.2, I notice that going into manual leveling backing out (with the return button) cause the following error:

Please see error: https://imgur.com/a/yQyCpsB

Perhaps something with my config but I don't know, haven't changed anything.

tititopher68, 2021-03-20:
It's a know Marlin issue.
Before flashing a New firmware, you must do a Reset settings then a store settings.
Ans After flashing this firmware, redo Reset settings, then store Settings and Restore Settings.

jjfs127, 2021-03-20:
> It's a know Marlin issue.
> Before flashing a New firmware, you must do a Reset settings then a store settings.
> And After flashing this firmware, redo Reset settings, then store Settings and Restore Settings.

So at this point, I need to reflash? I always do a reset once a reflash. I just tried Reset, store and restore - same error...

tititopher68, 2021-03-21:
BLTouch or Noprobe ?

jjfs127, 2021-03-21:
BLTouch

jjfs127, 2021-03-21:
> BLTouch or Noprobe ?

BLTouch

tititopher68, 2021-03-21:
Pre-compiled firmware or your own configuration ?
How is connected the BLTouch ?

jjfs127, 2021-03-21:
> Pre-compiled firmware or your own configuration ?
> How is connected the BLTouch ?

Not through the 5-pin JST connector but using the endstop 2-pin connector, I have this in the config:

#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

tititopher68, 2021-03-21:
Try with these files : https://drive.google.com/folderview?id=1ToN3jD23EUPoD-lPWe0p8O30Nlv0m4kK

tititopher68, 2021-03-21:
Try with these 2 config files : 
https://drive.google.com/folderview?id=1ToN3jD23EUPoD-lPWe0p8O30Nlv0m4kK

tititopher68, 2021-03-21:
I compliled with these config files : https://drive.google.com/file/d/1h2YFyoh9GPcaGVcX5UutcVw8aHWxdfYT/view?usp=drivesdk

tititopher68, 2021-03-21:
Don't forget to flash your LCD Screen too.
(with Jyers's DWIN_SET)

jjfs127, 2021-03-21:
> Don't forget to flash your LCD Screen too.
> (with Jyers's DWIN_SET)

Recently updated the DWIN_SET a couple of weeks back, can't remember now. Oh, I started from the "BLTouch-3x3-HighSpeed" config file, so I don't know...
But thanks for your config, I guess I can give it a go. Just wished I knew what was causing this.

Jyers, 2021-03-21:
Sorry I haven't gotten to this sooner, are you by chance running octoprint?

Alfagek, 2021-03-21:
I do have the same problem. Did flashed yesterday with version 1.2.2.
With autolevel i do get the error(M112) instant.
If i try it by octoprint, i do get the same. 
Did also try to do autolevel with out the usb, same problem.
If i try to print from SD i also do get instant M112. Octoprint wasn't connected to the ender.

All did with the precompiled version (5x5 fs).
So i did flashed version 1.2.0 to the ender, and now i dont get the error. 
Only "problem" i do have with that version is when i change the z-offset on printing, the printerhead is moving a few steps back and the goes on where he was. And now and then a "heating error" with a beep tone. Restart printer and now error anymore with "heating error" 

Ender 3 V2
Bltouch 3.1 (5 pin connector)
4.2.2 board

tititopher68, 2021-03-21:
I have this issue but not with this firmware. And i had just comment HS in Configuration_adv.h et uncomment #define DELAY_BEFORE_PROBING 100 in Configuration.h.

In our structure, we have 10 Ender-3 V2 with this firmware v1.2.2 and never had this issue. 
All BLTouch are connected by use with the Z-endstop connector. With 5-pins, there are some issues with Creality motherboard and BLTouch...

jjfs127, 2021-03-21:
> Sorry I haven't gotten to this sooner, are you by chance running octoprint?

No worries. Yes, running octoprint, seems like a recent change causing this. Been ok up to now.

Jyers, 2021-03-21:
Could you try this with your usb disconnected? Just to rule out the possibility of octoprint sending the M112 command.

jjfs127, 2021-03-21:
> Could you try this with your usb disconnected? Just to rule out the possibility of octoprint sending the M112 command.

Yes, tried it with the usb disconnected, and it didn't happen. Thoughts on why this is?

Jyers, 2021-03-21:
This means that something with the printer is causing octoprint to think that an error has occurred, and in response it sends an M112 command to make sure the printer shuts down. I believe this can be disabled in octoprint, though I don't know exactly what octoprint is seeing that puts it into an error state.

Alfagek, 2021-03-23:
> It's a know Marlin issue.
> Before flashing a New firmware, you must do a Reset settings then a store settings.
> And After flashing this firmware, redo Reset settings, then store Settings and Restore Settings.

This solved my problem indeed (5 pin connector bltouch)
Did this and a new flash.
2 days no M112.

Only i can't do a mesh bed from Octoprint (with bed visualizer). If i try that, i will get eventually the M112 on the screen.
So like @Jyers is all ready saying, there is something in Octoprint what send M112 to the printer.

jjfs127, 2021-03-23:
Thanks. 
After thinking about this. I suspect this has something to do with a recent change I made... 
I replaced the stock heatbreak with a bi-metal one and because the break is much shorter than the stock it had the adverse effect of essentially raising the nozzle height to the point where the tip of the bl touch (when retracted) just barely clears the nozzle's height from the bed. I think there's some checking going on here in this regard.
I either need to alter the Satsana shroud to raise the bl touch height or adjust the height of the heatbreak...

Jyers, 2021-03-24:
Closing this for now, If anything new comes up that indicates there might be a firmware bug, just update this and I can reopen the issue.

closed 2021-03-24 by Jyers

jjfs127, 2021-03-29:
I re-opened this is the hopes that would provide additional insight as the what the problem could be. At this point, I don't know what to make of it. Octoprint reports:
```
Recv: echo:Invalid mesh.
Recv: Error:Failed to enable Bed Leveling
Changing monitoring state from "Operational" to "Error: Failed to enable Bed Leveling"
Send: M112
Send: N2 M112*35
Send: N3 M104 T0 S0*34
Send: N4 M140 S0*97
Changing monitoring state from "Error: Failed to enable Bed Leveling" to "Offline (Error: Failed to enable Bed Leveling)"
Connection closed, closing down monitor
```
Any suggestion would be much appreciated. If not, you may close it once more.

Jyers, 2021-03-29:
Oh!!!!!! You meant backing out of the normal manual leveling menu... I thought you meant the mesh leveling menu! I absolutely know what is causing that, your ocoprint report helped a lot. I'll put out a fix soon.