Jyers/Marlin #320: M600 Filament change from Octodash does not work.

hoefs, 2021-03-25:
### Description

Filament change from Octodash does not work. This was working with the https://marlin.crc.id.au/ firmware files. I configured to use the M600 filament change (https://marlinfw.org/docs/gcode/M600.html). With your firmware files, nothing happens...

### Steps to Reproduce

1) Choose new filament color from filament manager on octodash.
2) Wait for filament to be removed (after hotend was heated)
3) ...

**Expected behavior:** [What you expect to happen]

Filament to be removed

**Actual behavior:** [What actually happens]

Nothing happens

tititopher68, 2021-03-25:
Hello, this is indeed normal in absolute terms.  Indeed the #define FILAMENT_CHANGE_UNLOAD_LENGTH parameter in Configuration_adv.h has been set to 0. Modify this parameter as well as #define FILAMENT_CHANGE_LOAD_LENGTH and recompile.  It should do it ....

Jyers, 2021-03-26:
This firmware makes no change to how M600 commands operate. You can try setting the load/unload distances in Control>Advanced, but besides that there should be no difference in the gcode execution

Jyers, 2021-03-27:
Any update on this?

hoefs, 2021-03-27:
> Any update on this?

Still trying to find out the settings that were default with the marlin.crc.id.au
Will flash it on Monday and check the settings as it worked flawlessly with that firmware for the V2 (4.2.2)

hoefs, 2021-03-28:
I measured my bowden tube from extruder to nozzle and it was about 450 mm. I have configured this in the menu with load and unload length + stored in eeprom. 
I have ran the change filament again from octodash and it worked perfectly.

closed 2021-03-28 by hoefs

Jyers, 2021-03-28:
Glad you got it working, enjoy the firmware!

hoefs, 2021-03-29:
> Glad you got it working, enjoy the firmware!

I am loving this firmware!
Btw the default in de configuration files is not 0. I dont know why you have changed them in your build?

https://github.com/MarlinFirmware/Configurations/blob/import-2.0.x/config/examples/Creality/Ender-3%20V2/Configuration_adv.h

```
/**
 * Advanced Pause
 * Experimental feature for filament change support and for parking the nozzle when paused.
 * Adds the GCode M600 for initiating filament change.
 * If PARK_HEAD_ON_PAUSE enabled, adds the GCode M125 to pause printing and park the nozzle.
 *
 * Requires an LCD display.
 * Requires NOZZLE_PARK_FEATURE.
 * This feature is required for the default FILAMENT_RUNOUT_SCRIPT.
 */
//#define ADVANCED_PAUSE_FEATURE
#if ENABLED(ADVANCED_PAUSE_FEATURE)
  #define PAUSE_PARK_RETRACT_FEEDRATE         60  // (mm/s) Initial retract feedrate.
  #define PAUSE_PARK_RETRACT_LENGTH            2  // (mm) Initial retract.
                                                  // This short retract is done immediately, before parking the nozzle.
  #define FILAMENT_CHANGE_UNLOAD_FEEDRATE     10  // (mm/s) Unload filament feedrate. This can be pretty fast.
  #define FILAMENT_CHANGE_UNLOAD_ACCEL        25  // (mm/s^2) Lower acceleration may allow a faster feedrate.
  #define FILAMENT_CHANGE_UNLOAD_LENGTH      100  // (mm) The length of filament for a complete unload.
                                                  //   For Bowden, the full length of the tube and nozzle.
                                                  //   For direct drive, the full length of the nozzle.
                                                  //   Set to 0 for manual unloading.
  #define FILAMENT_CHANGE_SLOW_LOAD_FEEDRATE   6  // (mm/s) Slow move when starting load.
  #define FILAMENT_CHANGE_SLOW_LOAD_LENGTH     0  // (mm) Slow length, to allow time to insert material.
                                                  // 0 to disable start loading and skip to fast load only
  #define FILAMENT_CHANGE_FAST_LOAD_FEEDRATE   6  // (mm/s) Load filament feedrate. This can be pretty fast.
  #define FILAMENT_CHANGE_FAST_LOAD_ACCEL     25  // (mm/s^2) Lower acceleration may allow a faster feedrate.
  #define FILAMENT_CHANGE_FAST_LOAD_LENGTH     0  // (mm) Load length of filament, from extruder gear to nozzle.
                                                  //   For Bowden, the full length of the tube and nozzle.
                                                  //   For direct drive, the full length of the nozzle.
  //#define ADVANCED_PAUSE_CONTINUOUS_PURGE       // Purge continuously up to the purge length until interrupted.
  #define ADVANCED_PAUSE_PURGE_FEEDRATE        3  // (mm/s) Extrude feedrate (after loading). Should be slower than load feedrate.
  #define ADVANCED_PAUSE_PURGE_LENGTH         50  // (mm) Length to extrude after loading.
                                                  //   Set to 0 for manual extrusion.
                                                  //   Filament can be extruded repeatedly from the Filament Change menu
                                                  //   until extrusion is consistent, and to purge old filament.
  #define ADVANCED_PAUSE_RESUME_PRIME          0  // (mm) Extra distance to prime nozzle after returning from park.
  //#define ADVANCED_PAUSE_FANS_PAUSE             // Turn off print-cooling fans while the machine is paused.

                                                  // Filament Unload does a Retract, Delay, and Purge first:
  #define FILAMENT_UNLOAD_PURGE_RETRACT       13  // (mm) Unload initial retract length.
  #define FILAMENT_UNLOAD_PURGE_DELAY       5000  // (ms) Delay for the filament to cool after retract.
  #define FILAMENT_UNLOAD_PURGE_LENGTH         8  // (mm) An unretract is done, then this length is purged.
  #define FILAMENT_UNLOAD_PURGE_FEEDRATE      25  // (mm/s) feedrate to purge before unload

  #define PAUSE_PARK_NOZZLE_TIMEOUT           45  // (seconds) Time limit before the nozzle is turned off for safety.
  #define FILAMENT_CHANGE_ALERT_BEEPS         10  // Number of alert beeps to play when a response is needed.
  #define PAUSE_PARK_NO_STEPPER_TIMEOUT           // Enable for XYZ steppers to stay powered on during filament change.

  //#define PARK_HEAD_ON_PAUSE                    // Park the nozzle during pause and filament change.
  //#define HOME_BEFORE_FILAMENT_CHANGE           // If needed, home before parking for filament change

  //#define FILAMENT_LOAD_UNLOAD_GCODES           // Add M701/M702 Load/Unload G-codes, plus Load/Unload in the LCD Prepare menu.
  //#define FILAMENT_UNLOAD_ALL_EXTRUDERS         // Allow M702 to unload all extruders above a minimum target temp (as set by M302)
#endif
```

Jyers, 2021-03-29:
Making it 0 is kind of a way to force people to realize that they need to set their distances. Especially if they have never used M600 before. Otherwise they might just assume its not supposed to retract all the way and keep using it like that.

hoefs, 2021-03-29:
> Making it 0 is kind of a way to force people to realize that they need to set their distances. Especially if they have never used M600 before. Otherwise they might just assume its not supposed to retract all the way and keep using it like that.

You are so correct. And in fact I have learned something new.