Jyers/Marlin #325: Z-offset while printing disturbs print

jsexauer, 2021-03-26:
While printing and using the Z-offset feature under the tune menu, if you adjust the number using the wheel, it frequently will corrupt the print job, at least while printing from a host.  Using the microstep up and down does not seem to cause this problem, only adjusting the number by 0.05 or more.  Running E3V2-ManualMesh-3x3-v4.2.2.bin.  Thanks!

Jyers, 2021-03-26:
I actually fixed this earlier today. I changed the wheel adjustment to also use an M290 command like the step up and down. This means it won't be accurate for large moves, but that shouldn't matter for small mid print offset adjustments. The fix will be in the next release.

closed 2021-03-26 by Jyers

jsexauer, 2021-03-27:
Awesome thank you