Jyers/Marlin #331: UBL View Menu Glitch when Viewer Show Values is checked

AngleV, 2021-03-26:
### Description

UBL View Menu is messed up when Viewer Show Values is checked and you press back during fill in action.

### Steps to Reproduce

Go to UBL View Menu and check the Viewer Show Values.
Now Select Mesh Viewer.
The firmware starts to fill in the screen with the mesh values. 
Press back before fill in action finishes. 
It goes back but the fill in action still continues to draw on the screen.

Jyers, 2021-03-27:
The fact that it lets you hit back before it is finished printing is astounding to me. I'll check the code to see what dark magic is allowing that.

MrDragon420, 2021-03-28:
Am encountering this is think as well

tititopher68, 2021-03-28:
Yes, exact, on my side I used a flag (activ), see image. to solve this issue.
I also set activ to 0 at the end of the "Set_Mesh_Viewer_status ()" function.

[image]

Jyers, 2021-03-29:
This should be fixed in the latest commit.

closed 2021-03-29 by Jyers