Jyers/Marlin #340: Installed BL Touch & Updated to BL Touch E3V2-BLTouch-3x3 - Printer says Power Loss issue.

VashBaldeus, 2021-03-27:
Installed BL Touch & Updated to BL Touch E3V2-BLTouch-3x3 - Printer says Power Loss issue.
Basically printer ceased printing claiming there's power loss issue, when there's no power loss at all!

Used PrisaSlicer to slice the gcode.

Video:
https://streamable.com/2u7ydo

Basically it's "looping" through that and does not really continue print.

Jyers, 2021-03-27:
Something is causing your printer to reboot, did this just happen randomly in the middle of your print?

VashBaldeus, 2021-03-27:
> 
> 
> Something is causing your printer to reboot, did this just happen randomly in the middle of your print?

Yes, randomly, I restarted the same print, trying to see if it happens again. And before this print after I installed the BL Touch I did print the CHEP Test cube and that went fine. Only change to the GCODE was 
M420 S1 ; Mesh Level
G29 ; Level Bed

added to the gcode.

Jyers, 2021-03-27:
Randomness implies a hardware fault. Perhaps a flakey connection or something of the like is causing your printer to reboot. Maybe something got knocked while you were installing the BL Touch? That would explain why it is only happening now.

VashBaldeus, 2021-03-27:
> 
> 
> Randomness implies a hardware fault. Perhaps a flakey connection or something of the like is causing your printer to reboot. Maybe something got knocked while you were installing the BL Touch? That would explain why it is only happening now.

restarted the print, might go through, will check if fails.

VashBaldeus, 2021-03-27:
Images: https://imgur.com/a/mwkub4c

Same print, same gcode, same spot it fails, I have a hunch it might be the print.

Jyers, 2021-03-27:
Very interesting. Check that gcode. If there are some incorrect commands or corrupter parts in there it might cause marlin to panic and restart.

VashBaldeus, 2021-03-27:
> 
> 
> Very interesting. Check that gcode. If there are some incorrect commands or corrupter parts in there it might cause marlin to panic and restart.

Well, only thing I've done to the gcode of the file after installing BLTouch was these two lines:
`
M420 S1
G29; Level Bed
`
As I had the file presliced before installing BL Touch.

VashBaldeus, 2021-03-27:
second attempt of printing same gcode, failed, I suspect gcode slice was bad, re-sliced.. lets see how this goes.

closed 2021-03-27 by VashBaldeus