Jyers/Marlin #353: Progress stuck on 0% (from OctoPrint with M73Progress Plugin installed)

ErnieBall, 2021-03-30:
### Description

When printing from OctoPrint, the progress remains at 0%.
I installed the M73 Progress Plugin and set the GCODE scripts as showed in the wiki page.

I am using the precompiled E3V2-UBL-BLTouch-15x15-HS-v4.2.2.bin (v1.2.2)

Here are the settings i'm using. Do i need to check "use time estimate" in M73 Progress Plugin?

[image]

[image]

tititopher68, 2021-03-31:
Just tested but, sorry, no issue...

ErnieBall, 2021-03-31:
I did another test print today and now the percentage shows up fine. Not sure what was wrong before.

Closing this issue.

closed 2021-03-31 by ErnieBall