Jyers/Marlin #358: Resume Print not clearing?

Legsmaniac, 2021-03-31:
### Description

Everytime I turn the printer on, I am presented with a box saying "Resume Print?  Looks like the last print was interupted - Conform or Cancel?"  Since my last print was in fact completed sucessfully, I have no idea why this is popping up.  So since I have nothing to resume, I selected Cancel and it's back to the start menu.  But if I switch the printer off and then on again, I'm back to square 1 with the box asking if I want to Resume Print.
So I tried selecting Continue instead and it starts heating up and the title at the top says Host Print.  I have no idea what Host Print is, it's certainly not any name of print file I've used.  I cancelled it and confirmed.  Back to normal start menu.
Switch off, switch on and the blasted Resume Print? box is back again!!!
Rinse and Repeat!

### Steps to Reproduce

I have no idea how to suggest how to reproduce this.  All I did was print a cookie cutter, finish and switch off.
Next thing I know was this happening next time I switched the printer on.

**Expected behavior:** [What you expect to happen]
I don't expect this to keep popping up when in fact I have nothing to resume!

**Actual behavior:** [What actually happens]
It keeps coming back and won't go away!

#### Additional Information

I have even flashed the latest firmware hoping this would clear it but no, it's STILL there!

[image]

tititopher68, 2021-03-31:
Hello, 

You must have a file PLR...(Power Loss Recovery).  After flashing a firmware, doing a Reset/Store and Restore Settings is usefull.
Delete this file and no more issue...

Are you also using Octoprint ? This file can appear randomly at the end of a print with Octoprint, like that without reason...

Legsmaniac, 2021-03-31:
Thank you for your reply.

Nope, there was no PLR file, just the last two gcode files of the last two prints I did.  
I deleted those, powered on and.....   Finally!!!!   Back to normal!

It's still very strange how it got to this stage in the first place and why it persisted until I was force to remove the gcode files.
If it happens again - I'll be back!

Jyers, 2021-04-01:
Interesting, as @tititopher68 said, this should only have happened if a PLR file was present. I'll close this for now, but feel free to update this if the issue comes back and I will open the issue back up.

closed 2021-04-01 by Jyers