Jyers/Marlin #359: Inexisting checkboxes

Regnak, 2021-04-01:
### Description
Hi,
When I want to check or uncheck "Filament sensor" or "Power-loss recovery", I cant see the checkbox like I'm seeing in JSTECH video.
When I long press in "Filament sensor" for exemple I listen a beep but I cant determinate if is check or not.
For information I take firmware named "E3V2-Noprobe-3x3-v422-v123c.bin" on Ender 3v2 stock with 4.2.2 mother board.

### Steps to Reproduce

I dont know reproduce that I update mother board first and screen next but with the screen I dont see like the JSTECH video annonce, bleu screen switch to orange, I just see loading marlin and next one more loading bar.

**Expected behavior:** Just see and interact with checkbox

**Actual behavior:** Cant see and interact with checkbox for "Filament sensor" or "Power-loss recovery"

#### Additional Information

[image]

tiagofreire-pt, 2021-04-01:
Re-reflash your LCD firmware with this one: https://github.com/Jyers/Marlin/tree/Ender_3_V2_Extensible_UI/Display%20firmware

Jyers, 2021-04-01:
Alternatively if you want to compile your own and don't want to flash the screen you can modify the configuration and comment out `#define CREALITY_DWIN_EXTUI_CUSTOM_ICONS`

This would obviously be without whatever modifications were made by Christophe

closed 2021-04-01 by Jyers