Jyers/Marlin #361: Filament Sensor is triggered but says no filament?

Sersch97, 2021-04-01:
I have used the E3V2-BLTouch-5x5-HS-v4.2.2 and the Bltouch works fine.
The problem is, that the Filament sensor is saying all the time that there is no filament in it. If i try M119 in the terminal it says that the sensor is triggered. As i understand it says that it should work properly, but it doesnt. 
When I start a print it levels and preheats. then it starts drawing the first line and right after that it goes to parking the nozzle and says filament runout.

Jyers, 2021-04-01:
The pin state of your sensor is most likely the opposite of what is set in the firmware. Unfortunately there is no way to make this adjustable in the on screen menu.  If you want to fix the issue you will need to compile your own version with this setting changed
`#define FIL_RUNOUT_STATE LOW` -> `#define FIL_RUNOUT_STATE HIGH`

closed 2021-04-01 by Jyers

defetzer, 2021-04-02:
Same here, but your workaround brings printer to work again.Thank you.