Jyers/Marlin #382: How to compile a slow variant?

mashkuov, 2021-04-04:
I'm having a problem with the HS versions only, and with the "slow" versions work just fine.
When I try to compile the code myself, I get the same problem as in the HS versions, so I assume that the default is the HS?

I have tried to change the value of Z_PROBE_FEEDRATE_FAST to match the value of Z_PROBE_FEEDRATE_SLOW, but it didn't help.
#define Z_PROBE_FEEDRATE_FAST (2*60)
#define Z_PROBE_FEEDRATE_SLOW (2*60)

Which part of the code do I need to comment or uncomment to compile a slow probe version?

aryajuanda, 2021-04-05:
there is an example configuration file for slow

https://github.com/Jyers/Marlin/tree/Ender_3_V2_Extensible_UI/Configuration%20Files/E3V2%20Templates/BLTouch-3x3

vhristenco, 2021-04-05:
Comment out the line  #define BLTOUCH_HS_MODE in Configuration_adv.h

closed 2021-04-05 by mashkuov