Jyers/Marlin #389: UBL Bed Leveling stops at "middle"

njmV, 2021-04-05:
Dear Jyers and all

First of all my congratulations for the wonderfull work you are doing with this FW!

Now regarding this "post".
I have compiled and used several times marlin.
This time I compiled latest source (today's) and changed some parameters from templates on folder E3V2 Templates related with UBL.
I changed GRID_MAX_POINTS_X  for 3 times with 3 different values.

### Description
First time GRID_MAX_POINTS_X to 15
Second time GRID_MAX_POINTS_X to 10
and latest GRID_MAX_POINTS_X to 5

### Steps to Reproduce
When I try to create a new mesh all of them stop at "middle", but always on same step for same GRID_MAX_POINTS_X
When steps GRID_MAX_POINTS_X is 10 always stops at 56
When GRID_MAX_POINTS_X is 5 always stops at 10

First when I set up GRID_MAX_POINTS_X to 15 I thought it was a memory failure. But with smaller GRID_MAX_POINTS_X, error keeps happening.

Googled arround for it and saw old posts for marlin regarding this defines:
#define PROBING_MARGIN 30
#define MESH_INSET 50 

but could not get it to work.

Might this be some bug?

If I can be usefull for some testing, please share steps.

Regards

[image]
[Config.zip](https://github.com/Jyers/Marlin/files/6260860/Config.zip)

tititopher68, 2021-04-05:
Hello,
The memory capacity does not allow storing more than 200 points, depending on the activated parameters (M503 returns that we can just store 2 Meshs of 100 points (10x10).
Then in UBL, there are real points, measured by the probe and interfered points. It depends on your Probe_X_Offset and Probe_Y_Offset. The head cannot go everywhere.
But if, as in my case, in UBL 10x10, with a Probe_X_Offset of -40 and a Probe_Y_Offset of -7, an X_MAX_POS set at 249, a PROBING_MARGIN set at 5 and finally a MESH_INSET set at 10, I get 91 points out of 100.
So to summarize, no issues ... Just adjustments to your settings to be made in your Configuration.h file

;-)

njmV, 2021-04-05:
Ok, understood! 
Thank you for your time clarifing and ultra fast response!

Will try find my correct parameters.

Meanwhile I could read some a manual fill the missing spots from https://github.com/Jyers/Marlin/discussions/387

Jyers, 2021-04-06:
As I just added to that discussion, you don't actually need to run any G29 P3 commands. The firmware does it for you after creating a mesh.

Jyers, 2021-04-06:
@tititopher68 From what I've gathered it seems to be best practice to keep the probing margin larger than the mesh inset, as that insures auto tilting will be inside of the mesh and not outside it. You do lose the perimeter of probe points, but those can be inferred. It also seems it would be better to have the mesh cover more of the bed rather than just being within the probes reach.

njmV, 2021-04-06:
Thank you very much Jyers.
So the G29 is done automatically by the FW. Will remove it from "my steps" :)!
Will also try to find the margin and mesh inset value within the "best practice" values: PROBING_MARGIN>MESH_INSET 

Less probes "gathered" but more mesh space cover.

I think I should close the topic? since it is all ok. Doing it and thank you all.

closed 2021-04-06 by njmV