Jyers/Marlin #396: Z-Offset Tool : Live Adjustement Issue

G-rom68, 2021-04-06:
### Description

Hi Everyone,

I didn't use a precompiled firmware, i compiled my own based on Jyers/Marlin Branch (sorry, i'm new to github, not sure about exact terms to use). i used the DWIN_SET as is. 
Everything works like a charm but the live adjustement setting in Z Offset Tool doesn't seems to work properly, i can modify the offset value with microstep UP and Down but the nozzle doesn't move.

### Steps to Reproduce

1. Select "Live Adjust" in Z-Offset Tool
2. click on "Microstep Up" or "Microstep Down" the Z-offset value change but the nozzle doesn't 

**Expected behavior:**  Clicking on "Microstep UP or Down" should not only change the Z-Offset value but also should have a "live" effect on the nozzle

**Actual behavior:** The Nozzle don't move when adjusting Z-offset value (with live adjust selected)

#### Additional Information
[Marlin-ZOffsetTool_Live_Adjustment_issue.zip](https://github.com/Jyers/Marlin/files/6267705/Marlin-ZOffsetTool_Live_Adjustment_issue.zip)

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Jyers, 2021-04-06:
Just to check that you know what you're looking for, the microstep items move the axis 0.01 mm. This corresponds to a rotation of the z axis stepper that is less than one degree. To notice the movement you either need to be very closely watching the Z axis coupler or paying very good attention to the nozzle.

G-rom68, 2021-04-06:
Thank you for reading my issue.
yes, but even when setting a "noticeable" movement (i.e few millimeters), no move is visible. but when i tried a Z Homing, i could check the new Z offset value (previously set) have been taken into account.

Jyers, 2021-04-06:
Interesting, so is the regarding only the Microstep buttons, or are you also seeing this issue with the main adjustment menu item?

G-rom68, 2021-04-06:
By "Main adjustment menu", you mean trough the menu where you can move X Axis/Y Axis/Z Axis ? If yes, the answer is No, i'm able to move X/Y/Z steppers

Jyers, 2021-04-06:
No, I mean in the z offset menu, you can also adjust the offset by clicking the z offset item and turning the knob, then clicking again to confirm. Does it move when you set a new offset in that way?

G-rom68, 2021-04-06:
Oh ok Sorry, no, no move when modifying Z offset Value. the only to see the effect of that modified value is to execute a Z homing.
So i was able to set my z offset but it's a bit annoying (and time consuming as this is the only way to set the Z offset)

tititopher68, 2021-04-06:
@G-rom and @Jyers, it's not a firmware issue and I don't see anything in the config files that can explain this.

I think you also have to look at the material side. I assume a BLTouch is used. Is the BLTouch connected to the dedicated 5-pin connector? Is the Z-Endstop switch still connected to the motherboard .... If so, you have to disconnect it (or move it as low as possible). If this switch is still connected, it always remains active, even with the BLTouch. For the Autohome this does not affect the operation but for the Z-Offset it does. If the nozzle can't go down then I think the switch is on and nothing will happen on the Z axis...

Jyers, 2021-04-07:
@tititopher68 But I can't see such issues impacting his printers ability to babystep or move the Z axis. It is certainly a strange issue though. @G-rom68, you are using the Extensible-UI Branch right? The older rewrite branch is very outdated and I've seen a few people get mixed up and use that instead.

xantrk, 2021-04-07:
I also had the issue on last week, self compiled, with microstepping. It did not appear to work, manual stepping woked fine and noticeably. Another issue I realized is that re-setting z offset to its current value, further reduces the offset. So assume I set it to -0.50, I just click again, and once again to set it to -0.50, nozzle moves further down. 

Not sure if that behaviour is changed with the updates last week, and I will not be near my printer ~12 hours, then I can test it again if needed.

Jyers, 2021-04-07:
@xantrk That issue was fixed last week, believe it or not you're the first person to notice it. Real head-scratcher actually. Turned out that the mesh offsets were getting re-applied every time the z axis tried to move. Fixed it by disabling the mesh when in the Z Offset menu.

G-rom68, 2021-04-07:
@tititopher68 & @Jyers : Yes, a BL touch is connected to the motherboard but my official BL touch is originally equipped with 2 connectors (A 3-pin one and a 2-pin one). I partially used the dedicated port (i plugged the 3-PIN connector only) on the motherboard and i also used the Z endstop port (i plugged the 2-PIN connector) the Z endstops switch have been deconnected and dismounted from the printer frame).
To be as precise as possible, my printer is a Ender3 V1 with a V4.2.7 motherboard plus an Ender3 V2 screen (everything worked fine before the installation of the new screen and this new firmware flash).

@Jyers : Yes i'm using the Extensible_UI Branch, i've compiled the firmware on March 10th, I don't know if there was a new version since ?

Jyers, 2021-04-07:
There have been two full releases since March 10th, so I'd certainly recommend updating.

G-rom68, 2021-04-07:
Oh yes, i'll took the last version. I will keep you updated once update done. 
Just to be sure, no need to update the screen firmware?

Jyers, 2021-04-07:
No, there are some new icons, but they are not necessary.

G-rom68, 2021-04-07:
I've just compiled a new firmware based on the new version and..... it works flawlessly !! 

Thank you for your time and for your advices !

closed 2021-04-07 by G-rom68