Jyers/Marlin #402: After printing, I can't turn the screen lighting back on

mosheas, 2021-04-07:
Bug - Ver 1.30 /Manuel mesh 3x3
When I turn off the screen lighting,and when printer done printing, I can't turn the screen lighting back on. Only turning off the printer and turned on, then the lighting back.

tititopher68, 2021-04-07:
I think just add this
ui.refresh_brightness();
in : void CrealityDWINClass::Draw_Print_confirm()

Edit : Just tested with this commit and it's ok...

Jyers, 2021-04-08:
Oops, yup print confirm is a weird edge case that I didn't consider. Fix will be out shortly.

Jyers, 2021-04-08:
Fixed, the next bugfix release will most likely come out in a day or two depending on how many issues get reported.

closed 2021-04-08 by Jyers

AngleV, 2021-04-08:
https://github.com/Jyers/Marlin/commit/12d37bd09bea9f5bb6d421ce2978a7b638a06767#commitcomment-49249090