Jyers/Marlin #403: 15x15 UBL BLTouch not probing all of the 225 points!

SD4LABS, 2021-04-07:
Hi, I hope you are doing well!
I haven't ever used GITHUB properly so I might not be doing it right, so sorry if it's not the correct way.

I’m using your “E3V2-UBL-BLTouch-15x15-HS-v4.2.2.bin” firmware from few weeks now. I updated to release V1.3.0 from V1.2.2 today.

In V1.2.2 from the beginning I had issue that whenever I created a new 15x15 UBL mesh, it wouldn’t probe all 225 points as it should, it instead used to probe 211 or 212 (one of the two randomly) out of 225 points and then gives option to go ahead and save the mesh. Another person mentioned similar issue in E3V2 group on FB but for him it probed even lesser points. It still used to print fine but just doesn’t probe all of the 225 points.

After updating to V1.3.0 now it probes 157 or 158 points (again randomly switching between one of the two whenever new mesh is created) out of 225 points.

I have tried both standard and HS firmware for both of the releases with exact same results.

tititopher68, 2021-04-07:
No issue...Please look at : https://github.com/Jyers/Marlin/issues/389

Jyers, 2021-04-07:
This is the correct behavior. UBL does not handle probing the same as bilinear leveling and such. Instead of the grid being based on where the probe can reach, the mesh is designed to cover the whole bed. This means that when probing, it can only reach some of those points. The rest will be inferred based on the probed points. This is why UBL has such large mesh sizes, to allow a very detailed mesh that covers the whole bed.

closed 2021-04-07 by Jyers