Jyers/Marlin #419: 1.3.0 10x10 UBL HS

Onejk1, 2021-04-09:
Hi

this now only probes 65 points? is that correct?

tommymonk, 2021-04-09:
Yes it's expected, depending on your nozzle/probe distance it will only be able to reach a certain amount of the coordinates, the rest will be automatically interpolated.

tititopher68, 2021-04-09:
No issue. Please look at : https://github.com/Jyers/Marlin/issues/403

Onejk1, 2021-04-09:
ok , still no explination why on V1.2.2 its would do 91 and now with same x and y offsets it only does 65

tititopher68, 2021-04-09:
Just because Probe Marging and MESH INSET are different in the configuration files (Configuration.h) and therefore in the precompiled firmwares between V1.2.2 and V1.3.0.
It is also included in closed issue # 389 (included in # 403)....

Jyers, 2021-04-09:
> ok , still no explination why on V1.2.2 its would do 91 and now with same x and y offsets it only does 65

The probing margin was adjusted to accommodate more setups. If you want more points you can always use 15x15 which gives you about 157. But 65 is still an 8x8 grid, so unless your bed has some serious skyscraper warps on it, you should be fine.

closed 2021-04-09 by Jyers