Jyers/Marlin #430: unable to compile with last commit

tome9111991, 2021-04-09:
hello
with last commit version i can't compile.
i use the BLtouch-3x3 config

got few errors:

Marlin\src\lcd\extui\lib\creality_dwin\creality_dwin.cpp: In member function 'void CrealityDWINClass::Save_Settings()':
Marlin\src\lcd\extui\lib\creality_dwin\creality_dwin.cpp:4411:19: error: 'struct CrealityDWINClass::EEPROM_Settings' has no member named 'tilt_grid_size'
   eeprom_settings.tilt_grid_size = ubl_conf.tilt_grid-1;
                   ^~~~~~~~~~~~~~
Marlin\src\lcd\extui\lib\creality_dwin\creality_dwin.cpp:4411:36: error: 'ubl_conf' was not declared in this scope
   eeprom_settings.tilt_grid_size = ubl_conf.tilt_grid-1;
                                    ^~~~~~~~
Marlin\src\lcd\extui\lib\creality_dwin\creality_dwin.cpp:4411:36: note: suggested alternative: 'ulong'
   eeprom_settings.tilt_grid_size = ubl_conf.tilt_grid-1;
                                    ^~~~~~~~
                                    ulong
Marlin\src\lcd\extui\lib\creality_dwin\creality_dwin.cpp: In member function 'void CrealityDWINClass::Load_Settings()':
Marlin\src\lcd\extui\lib\creality_dwin\creality_dwin.cpp:4415:3: error: 'ubl_conf' was not declared in this scope
   ubl_conf.tilt_grid = eeprom_settings.tilt_grid_size+1;
   ^~~~~~~~
Marlin\src\lcd\extui\lib\creality_dwin\creality_dwin.cpp:4415:3: note: suggested alternative: 'ulong'
   ubl_conf.tilt_grid = eeprom_settings.tilt_grid_size+1;
   ^~~~~~~~
   ulong
Marlin\src\lcd\extui\lib\creality_dwin\creality_dwin.cpp:4415:40: error: 'struct CrealityDWINClass::EEPROM_Settings' has no member named 'tilt_grid_size'
   ubl_conf.tilt_grid = eeprom_settings.tilt_grid_size+1;
                                        ^~~~~~~~~~~~~~
*** [.pio\build\STM32F103RET6_creality\src\src\lcd\extui\lib\creality_dwin\creality_dwin.cpp.o] Error 1
================================================================= [FAILED] Took 121.56 seconds =================================================================
Environment             Status    Duration
----------------------  --------  ------------
STM32F103RET6_creality  FAILED    00:02:01.557
============================================================ 1 failed, 0 succeeded in 00:02:01.557 ============================================================

tome9111991, 2021-04-09:
i downloaded last .zip from code
but the creality_dwin.cpp looks like a old version

tititopher68, 2021-04-09:
Just modify "void CrealityDWINClass::Save_Settings()" and "void CrealityDWINClass::Load_Settings()" in Marlin\src\lcd\extui\lib\creality_dwin\creality_dwin.cpp like that :

void CrealityDWINClass::Save_Settings() {
  **#if ENABLED(AUTO_BED_LEVELING_UBL)**
    eeprom_settings.tilt_grid_size = ubl_conf.tilt_grid-1;
  **#endif**
}

void CrealityDWINClass::Load_Settings() {
  **#if ENABLED(AUTO_BED_LEVELING_UBL)**
    ubl_conf.tilt_grid = eeprom_settings.tilt_grid_size+1;
  **#endif**
}

and it's OK.

jstio888, 2021-04-09:
Thanks, compiles now!

Jyers, 2021-04-09:
Fix committed

closed 2021-04-09 by Jyers