Jyers/Marlin #440: bl touch mesh error

drveni, 2021-04-10:
ender 3 v2 4.2.2 1.3... dont tested lower version

install bltouch antclabs . 

in g code i use (G28G29) and while leveling bed there is random error and interruption of process. no matter is it 3*3 or 5*5... its just halt. on printer got error M112 and... and i need to turn off-on and restart printer+octoprint server to work again. its all random. i cannot mesh bed. try both 3 or 5... HS version or normal. 

i download E3V2-BLTouch-3x3-v4.2.2.bin. flashed few times. im flashed and display and returun to stock and just bin.. its random crash. somehow, i put puncture in im PEI bed. because it wont level or get error. now im destroy my PEI bed whit puncuture. im on glass for now.   few times he puss Z offset so hard on glassas if bltouch does not exist,; on stock ?!!!? seems works ok. im not so long on stock bin, im on jyers. (microswitch Z is disconnected)

bed is leveled and is works ok when G29 core passes.. im not use M420 S1, because i remove bed and put glass, i change bed... and BLtouch is great for that purpose. 
is new bltouch from offical ender webpage, whit correct connector. connector are god wires, and all joint seems ok. 

if bedleveling pass ok, print is ok...(whit no interuptions)

any help? 

p.s. sorry for bad Eng.

tititopher68, 2021-04-10:
Hello,

I imagine your BLTouch is plugged into the dedicated 5 pin connector.
There are 2 ways to connect a BLTouch. The one using the dedicated connector or the one using the Z-endstop connector : https://e.pcloud.link/publink/show?code=XZD2ykZeSpKtAikhD7l5qDPO0cMpQc1lBl7

This method is mainly recommended. The connection on the dedicated port is subject to a random crash of the Hotend on the Bed. Why, we do not really know why, even if there are suspicions the mounting of the sensor support. The precompiled versions are all with the dedicated connector. It will therefore be necessary to modify the Configuration.h and Configuration_adv.h files and recompile.

drveni, 2021-04-10:
im use bltouch kit, 5pin connector because is cable whit connector and mainboard is same connector for bltouch. i saw that has clone, this is not clone. same as in video https://www.youtube.com/watch?v=NTwEaVxEe1s ,  https://www.youtube.com/watch?v=NDe6QXreBNs 

cables are not separeted for Z pin connector

im experiment whit "https://marlin.crc.id.au" for now i dont see problem for now.... i really like Jyers but wont work correctly

can i send some log files?? would that help? if yes please procedure because I don't understand coding...

Zaaap, 2021-04-10:
Does your probe stop deploying the pin and starts to blink red?
This happend to mine.
Solution was to take out the pin, using the black screw, and put a little bit of silicon oil on the thick end of it, using cotton swab.
But only a little bit, dry it with a tissue and put it back.
Since then it works great on my Ender 3 V2 with the dedicated port.

phenomeus, 2021-04-10:
same here. e3v2 1.3 (first time) and bltouch with 5pin dedicated connector. before I used smith3d firmware and OEM firmware. 

i modded the e3v2 also with Hero me gen 5 (first time) and use the compact bltouch holder there. 

if I auto home, everything works as expected but after ~1 min of idling its just falling down :/ no commands send or anything.

tititopher68, 2021-04-10:
Hello,
As I wrote above, there is really a problem with the dedicated connector and Marlin, regardless of the Creality or BTT motherboard or whatever.

The firmware is not in question, the heart of Marlin certainly and it is on the Github of Marlin that the problem must be posed.

It is really advisable to connect the BLTouch to the Z-endstop (for the connector, just cut it to separate the pins ...).
But if you want to keep this connection mode, you will be subject to this problem at random and in particular with Octoprint.  

Now some advice that has been given to me in the meantime : 
Uncomment these Lines in Configuration.h : 
 #define Z_MIN_PROBE_PB1
 #define DELAY_BEFORE_PROBING 200
Modify:
 #define Z_PROBE_LOW_POINT -20
And recompile everything...

gerdur, 2021-04-11:
Hello, exactly the same issue happens here with Ender3V2, Original BL-Touch3.1 connected to the dedicated 5pin on 4.2.7 mainboard and Jyers 1.30 5x5HS firmware. To make things 100% clear: 
Is is recommended to pull out the two pins (which ones?) off the dedicated 5pin mainboard connector and instead to wire these 2 pins to 2 pin Z-Limit switch on the mainboard (orientation is unclear, because one of these pins will have a pull-up resistor I guess). Is this understanding correct or I‘m completely wrong? Additionally the firmware has to be changed and recompiled? Is there a schematic of the 4.2.7 mainboard available on the web? Detailed instructions how to proceed would be appreciated! 
By the way: if it is NOT an issue of the v1.30 firmware, why didn‘t this happen with earlier versions of Jyers firmware? Thank you so much for clarification.

Xploder, 2021-04-11:
Check out this discussion

https://github.com/Jyers/Marlin/discussions/192

drveni, 2021-04-11:
ok, i'll reconnect, cut black-white and put on Z switch. i didnt know thats two tipes of wire... this are is from creality shop.

next question, i just download firmware precompiled firmware from https://github.com/Jyers/Marlin/releases and it should work. !? no need ressistor?

sorry for noob question.

and if i need to back on stock/originally  firmware .. just flash stock and no need to recconect wires again !?

drveni, 2021-04-11:
> Hello,
> As I wrote above, there is really a problem with the dedicated connector and Marlin, regardless of the Creality or BTT motherboard or whatever.
>
> The firmware is not in question, the heart of [...]
ok, i'll reconnect, cut black-white and put on Z switch. i didnt know thats two tipes of wire... this are is from creality shop.

next question, i just download firmware precompiled firmware from https://github.com/Jyers/Marlin/releases and it should work. !? no need ressistor?

sorry for noob question.

and if i need to back on stock/originally firmware .. just flash stock and no need to recconect wires again !?

drveni, 2021-04-11:
> Hello,
> As I wrote above, there is really a problem with the dedicated connector and Marlin, regardless of the Creality or BTT motherboard or whatever.
>
> The firmware is not in question, the heart of [...]
ok, i'll reconnect, cut black-white and put on Z switch. i didnt know thats two tipes of wire... this are is from creality shop.

ok ill just reconnect....and this should work?

next question, i just download firmware precompiled firmware from https://github.com/Jyers/Marlin/releases and it should work. !? no need ressistor? i dont need to recomplile anything?

sorry for noob question.

and if i need to back on stock/originally firmware .. just flash stock and no need to recconect wires again !?

Jyers, 2021-04-11:
> By the way: if it is NOT an issue of the v1.30 firmware, why didn‘t this happen with earlier versions of Jyers firmware? Thank you so much for clarification.

@gerdur I don't think this actually stems from the new version. The BL Touch has a lot of possible issues and such, so its nost likely coincidence that has cause this error to occur after the release.

That being said, there is one possible reason why this could be occurring after an update. Make sure you reset eeprom anytime you flash new firmware, as old eeprom settings can cause a multitude of strange issues, particularly for the probe it would seem. You can use the "Reset to Defaults" in the control menu to reset eeprom.

phenomeus, 2021-04-12:
so the only workaround is split the wires and wire it to Z stop if I don't want to recompile (not that advanced with 3d printer settings yet) with: 

`Now some advice that has been given to me in the meantime :
Uncomment these Lines in Configuration.h :
#define Z_MIN_PROBE_PB1
#define DELAY_BEFORE_PROBING 200
Modify:
#define Z_PROBE_LOW_POINT -20
And recompile everything...`

tititopher68, 2021-04-12:
If you connect the BLTouch with the Z-endstop, you must also modify other parameters accordingly in Configuration.h
 So recompile anyway.
 It is very important to do after flashing the firmware a Reset then Store and Restore Settings.

darco415, 2021-04-13:
Now some advice that has been given to me in the meantime :
Uncomment these Lines in Configuration.h :
#define Z_MIN_PROBE_PB1

What line number is this? I dont find it..

Xploder, 2021-04-13:
> Now some advice that has been given to me in the meantime :
> Uncomment these Lines in Configuration.h :
> #define Z_MIN_PROBE_PB1
> 
> What line number is this? I dont find it..

It doesn't exist and there's nothing in Marlin mentioning it, so it wouldn't do anything anyways.

Jyers, 2021-04-14:
Check the octoprint settings on the GitHub wiki for how to avoid random M112 errors. I'm closing this for now, if new information comes up that points to this being a firmware issue, update this thread and I can reopen the issue.

closed 2021-04-14 by Jyers

drveni, 2021-04-14:
> Now some advice that has been given to me in the meantime :
> Uncomment these Lines in Configuration.h :
> #define Z_MIN_PROBE_PB1
> 
> What line number is this? I dont find it..

im try to recompale, there is no line #define Z_MIN_PROBE_PB1..