Jyers/Marlin #446: UML 15x15  strucks on point 149 with version 1.3.0

Sherman-c, 2021-04-11:
Hi 
I encounter another strange issue with E3V2-UBL-BLTouch-15x15-v4.2.7.bin, at first, it seem ran ok, it completed with only 17x points were saved to eeprom.  Since the bed level test was unsuccessful, I cleared the mesh data, and attempted to recreate new one again, this time it struck on point 130, repeated to test point 149.  BLtouch issue?

Software 1.30 E3V2-UBL-BLTouch-15x15-v4.2.7.bin,
Bed level test starts Gcode
G28
M420 S1

Regards
Sherman

PS I'm going to test 10x10 UBL and see how it goes

Sherman-c, 2021-04-11:
[](url)

Sherman-c, 2021-04-11:
Sorry I pressed the wrong button

tititopher68, 2021-04-11:
Hello,

No issue
;-) just look at : https://github.com/Jyers/Marlin/issues/403

Sherman-c, 2021-04-11:
Thanks for clearing that up for me.

Cheers
Sherman

Jyers, 2021-04-11:
Sorry, I think @tititopher68 misunderstood. You definitely have a problem here. I looks to me like a probe failure of some kind. The most common cause of this would be a connection issue of some kind. If you check all the connectors and still face this issue, you should also try adjusting the set screw on the top of the BL Touch, as that is a crucial component and can easily be slightly too far in or out.

tititopher68, 2021-04-11:
Sorry, yes, @Jyers is absolutely right ... I am using a translator ... bad translation.

Sherman-c, 2021-04-12:
Hi Jake,

Thanks for the update, the problem seems gone away with UBL 10x10 without
touching anything else.  I hot glued the connector when I installed the
BLTouch.  Since I have problem with UBL, both 10x10 and 15x15 versions,  it
printed on air on certain lines near the center.  I switched to BLTouch
5x5, and it was working ok so far.  I will send you a video once I get my
raspberry pi box printed.

Cheers
Sherman

Jyers, 2021-04-14:
Closing this for now, feel free to update with any new information or questions. If there is any further info pointing to this being a firmware issue I will reopen the issue.

closed 2021-04-14 by Jyers