Jyers/Marlin #447: Certain Z-Offsets desync Display from M851

Xploder, 2021-04-11:
### Description

Certain values for Z-Offset show a difference between the display and M851.

### Steps to Reproduce

1. Set LCD to Z-Offset Z-2.09
2. Send M851 via Serial
3. Send M851 Z-2.09 via Serial
4. Check display

**Expected behavior:** [What you expect to happen]
Both values should match

**Actual behavior:** [What actually happens]
1. Set LCD to Z-Offset Z-2.09
[image]

2. Send M851 via Serial
[image]
**Output is Recv: Probe Offset X-43.00 Y-13.00 Z-2.10**

3. Send M851 Z-2.09 via Serial
[image]
4. Check display
**Display shows Z-Offset -2.08**
[image]

#### Additional Information
Same with Z-2.10, Z-2.11, Z-2.12
**In sync again at Z-2.13**

tititopher68, 2021-04-11:
Hello,
Do you have your own configuration's files (Configuration.h ans Configuration_adv.h) ? Because if you have uncomment "#define BABYSTEP_ZPROBE_OFFSET" in Configuration_adv.h, you will get what you describe. Do not uncomment this parameter.
Because M851 = Z-Offset + Babystep

Xploder, 2021-04-11:
No it's uncommented
[image]

tititopher68, 2021-04-11:
OK.
See this vidéo : https://youtu.be/fN_ndWvXGBQ
We can see a Z-Offset fixed at -1.17 and on the display -1.18 ... this seems to be normal behavior...

Jyers, 2021-04-11:
This is most likely a rounding issue of some kind. I can assure you the display is using the exact Z offset strait from the core marlin code. If you think this is an issue, I can look into making sure the value is rounded before it is displayed and that should create a little more consistency. Either way, I don't see this causing any major issues, besides some slight confusion.

Schnabulation, 2021-04-13:
Yo @Jyers regarding `#define BABYSTEP_ZPROBE_OFFSET` - what is your take on that? I had that thing uncommented as long as I have been self-compiling my firmware but @tititopher68 mentiones that I should disable it. Can I get your answer on that?

Jyers, 2021-04-13:
It is best to disable it for use with this firmware. The issue is `BABYSTEP_ZPROBE_OFFSET` changes the z offset whenever you execute a babystep, but this would mean babysteps can only be used to tune the z offset and not for leveling and such. For this reason I disabled it and coded it to modify the offset after babystepping when it is needed. If you have this uncommented you will most likely get double changes to the offset when babystepping.

closed 2021-04-13 by Jyers