Jyers/Marlin #454: Bed temperature won't zero

Timbo200, 2021-04-11:
I've been using Octoprint and really didn't want to get involved with custom firmware, but your firmware was just too tempting.  So far I've been very impressed.  Thanks!

Using 1.3.0 and E3V2-UBL-BLTouch-15x15-HS-v4.2.2 I noticed a small bug.

On the Control Menu after turning on the Bed Temperature you can't turn it off or set it back to 0.   It only goes to 5C.

I'd imagine this isn't too much of an issue, but I'm not sure if 0 here actually turn the heating full off which of course would be optimal.

Jyers, 2021-04-11:
Thank you for the reminder. Marlin changed something in their default config files a while ago and set the min bed temp to 5 for some reason. I'll fix this right away.

Jyers, 2021-04-11:
Easy enough, just pushed the fix, it will be in the next release.

closed 2021-04-11 by Jyers