Jyers/Marlin #462: Z offset height fade

Srizzyy, 2021-04-12:
### Description

When printing after probing the bed , the first layer seems very smooth and nice not close not far , perfect z offset , however at 4 or 5th layer , layers start to squish and it feels that the nozzle height gets closer so i have to raise the z offset from the screen manually during printing
### Steps to Reproduce
Start normal print with perfect nozzle height and wait till layer 5 or higher and watch the layers get squished and pressed hardly on the model

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]
Maintain the right nozzle position through the print completely and maybe add the z offset fade 

#### Additional Information
I’m printing using your 5x5 bl touch firmware. 

thanks J.

tititopher68, 2021-04-12:
In your StartGcode, After G28, you place M420 S1 Z(value). This value is the Z Fade Height.
By default without value or with just M420 S1 or G29, Fade Height is fixed to 10 by Marlin.

With Fade enabled, bed leveling correction is gradually reduced as the nozzle gets closer to the Fade height.  Above the Fade height no bed leveling compensation is applied at all, so movement is machine true.
 Set to 0 to disable fade, and leveling compensation will be fully applied to all layers of the print.

Srizzyy, 2021-04-12:
So do you say that the squishing issue after couple of layers is caused by the z fade ? or because of some other reason ? 
and if it is because of the z fade value , do i add the m420 s1 z(value) after g28 and before g29 or after g29 ?

Thanks

tititopher68, 2021-04-12:
It may be the reason, but not necessarily the only reason.  Other parameters such as the temperature of the bed and / or the nozzle, the characteristics of the filament (which can more or less retract when cooling, ...), etc ...
 If you are using an M420, do not use a G29 at the same time ... Do before your Bed leveling and in your StartGcode, add below G28, an M420 S1 Z0 for example to test ...

tititopher68, 2021-04-12:
But not a firmware issue

vaderag, 2021-04-12:
> ### Description
> When printing after probing the bed , the first layer seems very smooth and nice not close not far , perfect z offset , however at 4 or 5th layer , layers start to squish and it [...]

I had this too.
Ended up turning off mesh levelling and things are working like a dream now. 100% an issue

Srizzyy, 2021-04-12:
> I had this too.
> Ended up turning off mesh levelling and things are working like a dream now. 100% an issue

How is that an option when i have a warped bed ?

tititopher68, 2021-04-12:
@vaderag `I had this too.
Ended up turning off mesh levelling and things are working like a dream now. 100% an issue`

Sorry but, this is not a firmware issue. Why ? Because the Bed Leveling is carried out by a G29 command specific to Marlin, to all Marlin firmware. This means that your bed isn't warped, so no need for correction...

vaderag, 2021-04-12:
> @vaderag `I had this too. Ended up turning off mesh levelling and things are working like a dream now. 100% an issue`
>
> Sorry but, this is not a firmware issue. Why ? Because the Bed Leveling is [...]

See my thread here: https://github.com/Jyers/Marlin/discussions/212#discussioncomment-557737

The _only_ change I made was to switch out bed levelling, so bed levelling is not being handled correctly by the firmware. Whether that's Marlin or this I don't know, but it's 100% an issue

tititopher68, 2021-04-12:
The bed leveling works in the following way with the BLTouch (hoping that your parameters in Configuration.h and Configuration_adv.h are correct)
The sequence is started by a G29 command (Marlin) which will take the measurement of each point according to a grid defined in 9 points if 3x3, 25 points if 5x5, etc .....
the measurements are carried out by the BLTouch and directly recorded in the RAM of the motherboard. The firmware part of @Jyers does nothing, it just runs the command and that's it. Once the complete mesh is done, it is saved in the EEPROM always by a Marlin command ..... As for all Marlin-based firmwares.
Then it happens at the level of your Gcode of your print. By including M420 S1 Zvalue (command Marlin) after a G28 command, Marlin will use the Mesh in memory and make the Z axis go down or up very slightly to correct the faults as much as possible of the Bed on a defined Zvalue height .... and the firmware n has nothing to do with it ....
Then there are a lot of other things that can affect the print, starting with the preparation you need to do before a good bed leveling. You may find that the BLTouch is defective and provides incorrect information ....
But I won't insist any further ....

Jyers, 2021-04-12:
@vaderag Have you tried increasing or disabling fade height? If so it may be worth posting an issue on the main marlin repo and see what the devs have to say.

closed 2021-04-12 by Jyers

vaderag, 2021-04-12:
> @vaderag Have you tried increasing or disabling fade height? If so it may be worth posting an issue on the main marlin repo and see what the devs have to say.

Yeah, tried a few things. Honestly I'm getting perfect prints now without the mesh so I can't really be bothered messing any more!!