Jyers/Marlin #463: Disable extruder stepper with M18 E not working

Svahnen, 2021-04-12:
### Description

Trying to disable the extruder stepper to change filament mid print.
When sending the command M18 E nothing happens.

### Steps to Reproduce

1. Start printing something
2. Pause the print (I paused it on a specific layer with M0 in Cura when generating the gcode, but the same behavior applies when sending either M0 or M25 from OctoPrint terminal)
2. Send M18 E from the terminal in OctoPrint

**Expected behavior:** 
Extruder stepper should be disabled.

**Actual behavior:**
Nothing, the extruder stepper is still engaged.

#### Additional Information

Just sending M18 or M18 X works and will disable all steppers including the extruder, the same behavior can also be found when sending M17 and M17 X (the engage stepper command, its all or nothing).

Svahnen, 2021-04-12:
Im using E3V2-ManualMesh-3x3-v4.2.2.bin

tititopher68, 2021-04-12:
No firmware issue ;-) 
Juste because in Configuration_adv.h, you have 
by default : #define PAUSE_PARK_NO_STEPPER_TIMEOUT           // Enable for XYZ steppers to stay powered on during filament change..
Comment just this with // before #define...and it should be ok

Svahnen, 2021-04-12:
> No firmware issue ;-)
> Just because in Configuration_adv.h, you have
> by default : #define PAUSE_PARK_NO_STEPPER_TIMEOUT // Enable for XYZ steppers to stay powered on during filament change..
> Comment just this with // before #define...and it should be ok

Don't that mean that all steppers would automatically be disabled on pause? I only want the extruder to be disabled so that I wont accidently move anything.

Jyers, 2021-04-12:
This is an issue for the upstream marlin branch. This fork doesn't modify any aspects of gcode handling. If you try this on another firmware and it works, then we can talk, otherwise this should be posted as an issue on the main marlin GitHub.

closed 2021-04-12 by Jyers

Svahnen, 2021-04-12:
I have not tried this on other firmware, I only went by the official Marlin documentation and assumed it would work as stated there.
https://marlinfw.org/docs/gcode/M018.html