Jyers/Marlin #474: UBL 10x10 does not probe all points

funnybrum, 2021-04-12:
### Description
Probe is with offset -35, 0. Running G29 P1 and then G29 T1 produces the following result:

`NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN
NAN	0.025	0.009	0.007	-0.001	-0.013	-0.026	-0.019	-0.021	NAN
NAN	0.020	0.022	0.018	-0.001	0.008	-0.009	-0.010	-0.015	NAN
NAN	0.022	0.014	0.015	0.003	0.013	0.013	0.013	0.019	NAN
NAN	0.008	0.013	0.017	0.013	0.019	0.031	0.033	0.036	NAN
NAN	0.017	0.017	0.012	0.007	0.013	0.021	0.033	0.042	NAN
NAN	0.019	0.025	0.018	0.010	0.013	0.022	0.027	0.023	NAN
NAN	0.009	0.012	0.001	-0.004	-0.002	0.002	0.005	0.009	NAN
NAN	0.008	-0.003	-0.006	-0.026	-0.025	-0.026	-0.014	-0.015	NAN
NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN	NAN
ok`

Right column is expected to fail probing. The X_MAX_POS is 245 and to probe the right column it needs to get to 255. But all other points are reachable and should be probed.

### Steps to Reproduce

1. Set probe offset to (-35, 0)
2. Go to Level menu
3. Select "Create new mesh"

**Expected behavior:** Probe all points except the one on the right column. Or total of 90 points.

**Actual behavior:** After probing 64 points the probe process completes.

Jyers, 2021-04-13:
This is the correct behavior. The probe is also not able to reach a perimeter of points around the bed. If you compile your own and don't think your probe will hit your bed clips, you can change the PROBING_MARGIN to match MESH_INSET.

closed 2021-04-13 by Jyers

funnybrum, 2021-04-14:
Didn't know about the PROBING_MARGIN. Now it makes sense. Thank you for the shared details!