Jyers/Marlin #486: Max extruder Temp cannot be set higher than 260

mattranieri, 2021-04-14:
Hi, since the 1.3 update i cannot set my extruder temp higher than 260.
this is an issue for me as i print ASA at 265.
is this something that can be fixed via the console?

Mineplayerminer, 2021-04-14:
I have the stock firmware 1.0.2 and I can go only up to 260. There even may be the limits of the extruder temperature.

tititopher68, 2021-04-14:
Exactly, the maximum usable value is 260⁰C.  The theoretical maximum value is 275⁰C but thermal protection (overshoot) has been set at 15⁰C (see Configuration.h).  Also the maximum usable temperature is 275⁰C-15⁰C = 260⁰C.  You can change the value of the Overshoot, but at your own risk for your printer ....

mattranieri, 2021-04-14:
For those of us rocking an all metal hot end with upgraded cooling this isn't really an issue. I appreciate the need for runaway protection however nylon or polycarbonate may need higher temps.

tititopher68, 2021-04-14:
Then you need to modify your Configuration.h file according to your thermistor ... The parameters given are for an Ender3V2 with an original Hotend...

Jyers, 2021-04-14:
As @tititopher68, this is a safety feature defined by marlin. If you would like a higher max temp you may want to look into compiling your own version so you can modify anything you need.

closed 2021-04-14 by Jyers