Jyers/Marlin #492: Z axis not rising at the start of Homing and probing loop during meshing with BlTouch

Bangber, 2021-04-14:
This happened continuously, but on 2 occasions. 

On my first encounter, I was using version 1.2.2 BlTouch HS 15x15 for 4.2.2, and i was upgrading my printer into a dual Z axis with single stepper and pulleys. The only thing i was messing with is the screw and adjusting the Z stepper position to fit the top pulley, and yet i haven't even installed my 2nd screw nor do i even mess with plugging/unplugging any cable. 

But when i try to use my printer, i noticed something strange when homing for the first time after turning the printer on, since **the Z axis rises, then stop for a while**, then proceed the homing. But when trying to home from then on, the **Z will never rise at the beginning**, but will rise extra (stop and start after the probing) at the end of homing (with a distance of about 5mm but doesn't update its position value displayed on the UI, so if i set my Z-offset to 0 then it's still 10mm). This also happen even when trying to start meshing, where there will be auto-homing. This issue "resets" itself after resetting my printer, so the Z will rise at the beginning of first homing n stall, then wont at the next attempts.

What's even worse, during meshing, there will occur a **probing loop** on the same point which seemingly never end. The point where this occurs differs on every attempt, one at 42th, other at around 30 or 50. The probe counter will still rise after every 2 probes. The bed is 75C, hotend 150 (i was trying to print PETG).

I tried updating to 1.3.0 HS, no luck. 1.3.0 not HS, still no luck. Then i tried resetting all my setting, deleting my eeprom from sd card, then ejecting the sd card before booting, still no luck. Then i tried lowering my bed to 60, no luck, but i remembered lowering to 50 (which i never had since even on PLA i use 60) and deleting my mesh then restart, if i recalled correctly that solves it (or was it more than one attempt). After that, i restarted, raise my bed to 75, and everything is good.

That was 2 days ago, and i managed to use the printer to do 2 prints on different days.

Today, i tried reworking my Z screw again by pulling it off the stepper and nut as how i did 2 days ago, and then i encountered the same problem again, but this time the solution i used didn't solve it.

I have huge suspicion that this might be a hardware issue since a firmware update doesnt help, but i just can't see the chance of it happening due to the odd behaviours of the error.

I documented my error from my 1st day of encounter, and didnt bother to do it again for my 2nd day since it's the exact same problem.

Bangber, 2021-04-14:
I forgot to post my documentations:

https://user-images.githubusercontent.com/72666518/114707202-2516d880-9d54-11eb-9edb-0348b8c6cf57.mp4

https://user-images.githubusercontent.com/72666518/114707167-1af4da00-9d54-11eb-8572-58f9827f1c6f.mp4

If it is actually a hardware problem, a help would be very appreciated. Thanks!

Bangber, 2021-04-14:
Can confirm that when I tried going back to creality's software, it works perfectly (the homing, at least, since there isn't any UBL). Going back to this firmware the problem still occurs

tititopher68, 2021-04-14:
Hello,

Probing loop: It does look like a hardware problem. Motherboard, BLTouch or cable (cable itself, connectors, etc, ?)
Are you connected to the dedicated port ? I read more and more issues with this port... It is preferable to connect the BLTouch via the Z-Endstop (requires modifying Configuration.h and re-compiling).

Bangber, 2021-04-14:
Hi, thanks for the reply!

I'll try checking my motherboard and its cables. Yes i'm using the dedicated port. Do i have to buy an adaptor if i am to use the Z-Endstop one? Also, please explain to me where can i get the Configuration.h because I am so clueless when it comes to compiling the firmware on my own.

I could understand that it's my bltouch error by only looking at the probing loop and the fact that the error persist after updates, but what i'm still not so sure is why does the Z acts that way during homing, and the fact that this suddenly occurs after i tried modifying my z leadscrew, twice, and that it doesn't seem to occur on other firmwares (without UBL). I also dont have octoprint nor a rasp pi connecting to my printer, just wanted to put that out here.

Jyers, 2021-04-14:
My best guess regarding why this might not occur on other firmware's would be that perhaps this is a marlin issue of some kind specifically with UBL, but theres really no good way to know for sure. I'll keep an eye on the upstream branch for any mention of such bugs.

tititopher68, 2021-04-14:
Regarding Z-Homing, it seems normal to me. When the printer is powered on, the x, y, z positions are not known. Also according to Marlin, with the BLTouch and the UBL (why only for the UBL, I don't know ...), he raises the z axis for safety then does his Z-Homing. Then, once the good positions have been acquired and the axis is at the desired minimum height, the z axis is no longer raised ...
On the other hand, your way of defining your Z-Offset does not seem correct.

For connection to the Z-Endstop:
[image]

tititopher68, 2021-04-14:
@Bangber, I'm also on BLT + UBL ... and my Z-Offset (-1.63) is OK without having to correct it. I also have no problem with Bed Leveling, but I am connected via the Z-Endstop.

Bangber, 2021-04-14:
Some interesting updates: i checked my cables and board, doesnt seem to have anything damaged. replug my bltouch cable (still on the dedicated port), tried running v1.3.0 n still no luck. And just now, i went back to version 1.2.2 and the homing works normally. Tried to go back to 1.3.0, error persists. Back to 1.2.2 again, and the error disappears. But, the issue with the probing loop is still there in 1.2.2.

Regarding the Z, i realized the Z will move extra not only during end of Z, but even when i commanded the Z axis to move in the prepare setting. I havent tried to see what happens during printing. This issue is exclusive to 1.3.0.. I'm quite sure this is not intended.

Also, how do you have the 2 cables to connect there perfectly? do you use a new cable for the bltouch, or do you splice the current one with the Z endstop cable?

Jyers, 2021-04-14:
Some BL Touch kits come with a cable that has two connectors, while the newer ones use one for the dedicated port. This is why the default is the port as not all kits support it. You can cut the connector to do it, but I always wonder if there is a better way. I run a BL Touch on a 4.2.2 board with absolutely no issues whatsoever.

Jyers, 2021-04-14:
I'll take a look into the config changes to see if maybe something changes with the Z raise. Most likely this is a bug carried over from the upstream marlin branch as I think I merged between 1.2.2 and 1.3.0. I'll check the latest fixes to see if its mentioned.

Bangber, 2021-04-14:
Another interesting update: i tried the 3x3 HS version 1.3.0, and when leveling it only probes 3 points from the first row, and it says leveling complete. Not sure if this intended, or if it might be a clue to what is causing my errors

Jyers, 2021-04-14:
Reset eeprom and try that again, definitely not what should be happening

Bangber, 2021-04-14:
I did reset it, and it still does that. Any idea to what's causing it? do you think i should try plugging my bltouch to the z endstop port? Not sure if i want to splice a (2) perfectly good cable(s) but if that solves it then it's worth it since right now i can barely use my printer.

Also, i noticed when moving the Z axis, it stutters almost every few steps, and it thinks that it reaches the determined height even though it hasnt. i can confirm this because when i tried to move the Z from "160" back to 50 after it stutters to raise, it just keeps on going and hits my bed. i tried to move the X too, and it delays after the input. the stuttering and doesn't always occur after printer restart, and it's just so confusing to why. starting to think that maybe it has something to do with my board?

Jyers, 2021-04-14:
Oh wow, yeah that definitely seems like a serious hardware issue of some kind. Just a sanity check, you're sure your using the correct version for your board (4.2.2 or 4.2.7)

JP-rg, 2021-04-14:
Just another note, from someone having similar problems, have measured voltages and come to same conclusion as to dedicated port missing resistor. (Mentioned by Xplorer in discussions) For info it doesn't matter what firmware I use the problem is there. Ender3 V2, latest Creality packaged BLTouch> Am about to rewire and compile to check. I will report back.
If it works with further checking I think I will solder 10k resistor on the back of the board.

BTW I have contacted Creality support but as of now no reply.

Jyers, 2021-04-14:
@JP-rg Just out of curiosity, how old is your V2? I just find it strange that I'm able to run a BL Touch on the port with no problems at all while other face fatal errors. Perhaps creality has fixed the missing resistor on newer boards.

JP-rg, 2021-04-14:
My V2 is only a month old, but delivered with 4.2.2 mother board. Can't seem to find date actually made...

Bangber, 2021-04-15:
So now the probing works again, and the Z no longer move extra steps. What i did was initiate creating mesh with the bed on 0 temp (i remembered doing something similar on my first encounter and it fixed it). BUT the mesh result was EXTREMELY inaccurate, where it says there is s 4mm difference on the bed height. When i tried to print, the nozzle hits my bed, damaging it. I assume it's an inaccuracy on the Z axis since there seem to be a pattern or some sort where the more it probes, the larger its reading is.

It's the correct 4.2.2 version of the 1.3.0, and i have 0 idea on why this happens. Tried checking the cable on my z stepper (since i suspect my tinkering have something to do with this) and yet after replugging it, it doesn't seem to do anything.

@JP-rg please do update on your attempt, i wonder if that's the cause of it. But for now i'm just so confused on what to do

Jyers, 2021-04-15:
@Bangber So sorry to hear that you're still facing issue. I wish I could help more, but from what I can tell this all comes down to either creality's faulty hardware, or bugs from upstream marlin. All of which are out of my hands. Hopefully together we can track down the real source of this issues so we can finally have a clear solution rather than all this confusion.

Bangber, 2021-04-15:
@Jyers yeah, thanks a lot for your attention! Love the firmware while it lasted, hopefully i can get a way to get it running again soon

JP-rg, 2021-04-15:
@Jyers Your Firmware is fine (apart from either my settings or something else in UBL leveling), it’s a motherboard issue…… I went with the 10k resistor option to solve the problems. Photo attached 
[image]

I have tested both 3x3 and 5x5 bilateral pre-compiled copies plus UBL 10x10 and 15x15 pre-compiled copies of your Firmware. They work as follows:

No problems on 3x3 on firmware version BLTouch-3x3-v4.2.2.bin
No problems on 5x5 on firmware version BLTouch-5x5-v4.2.2.bin
Consistently Stopping at probe 65 of 100 on firmware version E3V2-UBL-BLTouch-10x10-v4.2.2.bin
Consistently Stopping at probe 157 of 255 on firmware version E3V2-UBL-BLTouch-15x15-v4.2.2.bin

I wonder if it’s my settings somewhere.. 

Aside from the above, the probing is functioning consistently  and not producing the random errors that were persistent before.

As to errors while running Octoprint, they are as follows:

No problems on 3x3 on firmware version BLTouch-3x3-v4.2.2.bin
No problems on 5x5 on firmware version BLTouch-5x5-v4.2.2.bin

Consistently Stopping at probe 65 of 100 on firmware version E3V2-UBL-BLTouch-10x10-v4.2.2.bin with the following message 

Bed Visualizer Error
Timeout occured before processing completed. Processing may still be running or there may be a configuration error

Consistently Stopping at probe 157 of 255 on firmware version E3V2-UBL-BLTouch-15x15-v4.2.2.bin
with the following message 

Bed Visualizer Error
Timeout occured before processing completed. Processing may still be running or there may be a configuration error

Jyers, 2021-04-15:
The stopping at 65 and 157 "issue" has been well documented and just relates to how UBL handles the mesh. See #403 for more info. The octoprint timeout is most likely an issue with its settings and how long its set to wait before throwing an error. I would recommend using M420 V1 instead of re-probing the bed in your bed visualizer script.

JP-rg, 2021-04-15:
Many thanks for the Other issue, I will look into it when I get a moment.

I hope the testing shows that it's a Creality hardware problem that is solved with 10k resistor, what a pain..... how many people have a soldering iron, 10k resistor and the NERVE......

gothcha, 2021-04-15:
It is good to see further confirmation that the 10K resistor solves the main unreliability issue with the dedicated port. I am running later version of 1.2.2 with the upstream merge, but prior to the brightness mods, and lime Jyers have had no issues with UBL probing.

Bangber, 2021-04-16:
@JP-rg Gotcha, thanks a lot for the update! Although I'm not sure if it's because yours is bought recently (post-november version with the layer shift issue) or inherently it's a flaw found in all units. Hope someone with pre-November unit can confirm this too. May i know what's the ideal current between both pins so that i can check it for myself?

Also, for those who have the same issue with the probing and wants a quick fix, I've been consistently succeeding in creating a mesh by making one with bed and hotend temp of 0, save mesh, THEN do the real meshing with your prefered temperature. I'm going to suspect that since this issue surfaces for me during my prints of PETG when needing bed temp of 75C, it's because the board can't handle that heat during probing. Resetting EEPROM doesnt seem to help, but at least there's a reliable fix for this.

My probing value is now working accurately. I suspect the reason to the inaccuracy that causes my bed to be hit is because of a slack in the Z stepper coupler since after i tighten it up and redo the meshing, i have realistic results which shows when I print. Might also be an actual error since i dont think my coupler is even slacking, but for now I'm going to conclude that that's the cause.

Jyers, 2021-04-16:
Closing this for now as it would seem its not a firmware issue. Please feel free to keep updating this or move it to a discussion if you want to further investigate the issue.

closed 2021-04-16 by Jyers