Jyers/Marlin #493: Problems with the Icons and custom images

Mineplayerminer, 2021-04-14:
### Description

I uploaded custom icons and start image on the LCD and It shows some kind of noise non-sense.

### Steps to Reproduce

I extracted the 9.ICO
Edited the images using paint.NET and saved it into its original quality
I compressed the images back into 9.ICO
And flashed the DWIN_SET on the LCD

**Expected behavior:** The icons and images should have applied.

**Actual behavior:** The icons were black and the image was just a noise.

#### Additional Information

These are the images how they looked like after editing all of them.
[image]

This is the settings the paint.net gave me when saving the images.
[image]

This is, how it looked after the flashing:
[image]
[image]

This is not how it supposed to look like.

I have formatted the SD card correctly as it

Giadej, 2021-04-14:
What tool are you using to recreate the 9.ico file?

I've used a chinese software (named 迪文ICO V3.6) to rebuild iconset from jpg and works fine

[image]

Giadej, 2021-04-14:
What tool are you using to recreate the 9.ico file?

I've use chinese tool (named 迪文ICO V3.6.exe) and works fine
[image]

P.S.: make sure that the size of the 9.ico file is close to the original

Mineplayerminer, 2021-04-14:
> What tool are you using to recreate the 9.ico file?
> 
> I've use chinese tool (named 迪文ICO V3.6.exe) and works fine
> [image]
> 
> P.S.: make sure that the size of the 9.ico file is close to the original

Im using dwin-ico-tools-main as it works completely fine. https://github.com/b-pub/dwin-ico-tools

Giadej, 2021-04-14:
I can't get the splitIco.py tool to work, when I run it it creates a new folder but inside I find only the DWIN_ICO.cpython-38.pyc file and not the extracted any icons.

Mineplayerminer, 2021-04-14:
> I can't get the splitIco.py tool to work, when I run it it creates a new folder but inside I find only the DWIN_ICO.cpython-38.pyc file and not the extracted any icons.

Did you open cmd in the right directory?

Giadej, 2021-04-14:
i have the tool in C:/ico/ and use command C:/ico/splitIco.py 9.ico

(Chinese tool works well)

Mineplayerminer, 2021-04-14:
> i have the tool in C:/ico/ and use command C:/ico/splitIco.py 9.ico
> 
> (Chinese tool works well)

Very sketchy to make it on C: and a single folder 😅

Giadej, 2021-04-14:
> > i have the tool in C:/ico/ and use command C:/ico/splitIco.py 9.ico
> > (Chinese tool works well)
> 
> Very sketchy to make it on C: and a single folder 😅

😅
Could you share the steps you followed to get the script to work?

Mineplayerminer, 2021-04-14:
> > > i have the tool in C:/ico/ and use command C:/ico/splitIco.py 9.ico
> > > (Chinese tool works well)
> > 
> > 
> > Very sketchy to make it on C: and a single folder 😅
> 
> 😅
> Could you share the steps you followed to get the script to work?

I will send you a video. Ill make it now.

gokhanselen, 2021-04-14:
Hello. What should I do to make the menu beautiful?

hayhay7789, 2021-04-15:
Having a similar problem. I got the boot screen working ok, and some of the icons work, but the rest are blank or scrambled. 

I’ve used GIMP to edit the hue of my images (spare the Creality logo and the loading bar) then opened each one with MS Paint and saved to disk to convert them to baseline. I also used the python tools which seem to work ok.

Mineplayerminer, 2021-04-15:
> Having a similar problem. I got the boot screen working ok, and some of the icons work, but the rest are blank or scrambled.
>
> I’ve used GIMP to edit the hue of my images (spare the Creality logo and [...]

I have a similar problem, I dont have the Info button, and the Cancel/Confirm buttons.

Giadej, 2021-04-15:
> I will send you a video. Ill make it now.

Thank you, if you have had problems with the video you could also describe the steps of "how to"...

Mineplayerminer, 2021-04-15:
> > I will send you a video. Ill make it now.
> 
> Thank you, if you have had problems with the video you could also describe the steps of "how to"...

Extracting 9.ICO, python/or the directory and .exe of "python splitIco.py 9.ICO output"

Reverse: python makeIco.py output 9.ico

("output" is the name of the folder)
You also need a pillow library: python -m pip install pillow

Giadej, 2021-04-15:
> > > I will send you a video. Ill make it now.
> > 
> > 
> > Thank you, if you 
> You also need a pillow library: python -m pip install pillow

Solved, thank you

Mineplayerminer, 2021-04-15:
The issue has been resolved. You need to be careful when saving/exporting the .jpg image.

In GIMP, you need to uncheck everything except the keep original quality. You can set it on 100% if you're starting from scratch
[image]

Most of the programs, don't have this feature.

closed 2021-04-15 by Mineplayerminer