Jyers/Marlin #494: Filament run out

menthol1960, 2021-04-14:
installed firmware 

set everything up smooth as silk works perfect except one thing 

Filament run out sensor enabled if fw meter checked switch etc etc 

started a little print cut filament and it carried on even after filament past through sensor tried optical and analog switch still same any help would be appriciated

Jyers, 2021-04-14:
I would check the switch connection as the firmware doesn't change anything about how filament runout is handled.

Mineplayerminer, 2021-04-14:
> I would check the switch connection as the firmware doesn't change anything about how filament runout is handled.

Heyy, I want to know how can I add a brightness option in my Ender 3 V2 LCD without any other additions. Can you help me, please? Jonny11#8591

menthol1960, 2021-04-14:
> 
> 
> I would check the switch connection as the firmware doesn't change anything about how filament runout is handled.

I metered the connection manually and when i pull filament out it triggers switch

i have it enabled in the FW via panel but when i test it carries on printing even after filament pulled out ?

what am i missing ?

tititopher68, 2021-04-14:
And do a Reset to defaults then Store Settings and Restore Settings ...
and also check the operating mode of the sensor (LOW or HIGH mode) with a multimeter for example by making a continuity test between the VCC terminal and Signal ...
I think it works in HIGH mode, so it will be necessary to change the parameter LOW to HIGH for #define FIL_RUNOUT_STATE in Configuration.h and recompile ...

menthol1960, 2021-04-14:
enabled everything still the same starts to print and then parks with message on screen saying filament run out detected and parks the head .

even though filament in device ??

menthol1960, 2021-04-14:
ITS ok folks i sussed it was a faulty sensor

menthol1960, 2021-04-14:
put on switch sensor works perfect

closed 2021-04-14 by menthol1960

Jyers, 2021-04-14:
If it is parking right away it might be due to your original sensor being inverted as to the default set in the firmware. https://github.com/Jyers/Marlin/discussions/481 is discussing a possible switch, but I'm glad you have a second one that works properly.