Jyers/Marlin #495: Filenames not visible or truncated

Nadeox1, 2021-04-14:
Hello,

I installed 1.3.0 and noticed that some of the filenames are not showing up fully, or at all.
See the attached images:

In this, the _Cable Clip_ filename is truncated, and the 3rd gcode filename is not showing up.
[image]

Same thing here. If I select the blank one for printing, the name appears correctly on the print view.
[image]

Could it be an issue with long names? I usually name them so to include Print Time, Layer Height and Grams. Worked fine with the previous version of the firmware.

Mineplayerminer, 2021-04-14:
If the file has more than 32 characters in the name, it may not show correctly.

Jyers, 2021-04-14:
I believe if they are not showing up they are longer than the firmware can handle. The ones that get truncated are most likely due to putting a '.' in the name, as to remove the file extension the firmware crops everything after the '.' I'll put this on my list to see if I can change how this is handled, but for now I'd avoid super long filenames or periods.

closed 2021-04-14 by Jyers

Nadeox1, 2021-04-15:
Ok, thanks for checking! Using shorter names works fine.
I just wanted to add that in the previous version (when this was still called _Ender V2 Rewrite_) this issue was not present. Perhaps it got introduced with some update. Anyhow, great work as usual :)

Jyers, 2021-04-15:
That's interesting as I believe it uses the same framework. Maybe I'll take a look back and see what's changed. Thanks for letting me know!