Jyers/Marlin #500: Printer resets on issuing G29 S0 to view levelling mesh (in octoprint)

Hraefin, 2021-04-14:
I am running the precompiled firmware E3V2-ManualMesh-5x5-v4.2.2.bin from the 1.3.0 release.
Everything works perfectly except when I issue a G29 S0 command (via octoprint terminal).

When I issue this the printer does a reset and octoprint says the firmware reported an error.  Very vague.
I know this might be an issue with octoprint and not the firmware, but I do not know of another way to verify whether or not this occurs with a different terminal.  Having said that, it worked fine with the vanilla firmware.  My end-goal was to verify the mesh wrote to the non-volatile memory properly and also to visualize the bed levelling mesh.

If I can do any steps to troubleshoot this, please let me know and I will gladly do what I can.

tititopher68, 2021-04-14:
Hello,

For this firmware, itsn't G29 S0 but M420 V ! Because Manual Mesh don't use MBL mode but Bilinear Mode ;-)

Jyers, 2021-04-15:
@tititopher68 is correct, manual mesh uses bilinear leveling with the probe manually option. M420 V1 will do what you want

closed 2021-04-15 by Jyers

Hraefin, 2021-04-15:
Thank you folks!  I was not aware of the M420 V command and it worked exactly as expected.  I appreciate your time.  I do wonder why the printer would do a hard reset when the G29 command was issued though.

Awesome work on this firmware.  It's amazing what you're doing with the UI and functionality over the vanilla marlin firmware.

Jyers, 2021-04-15:
In case you're interested, as @tititopher68 said, the S parameter of G29 is to set the xy probing speed, hence why setting it to zero causes a firmware error. Check out this page for more info on the bilinear gcode: https://marlinfw.org/docs/gcode/G029-abl-bilinear.html