Jyers/Marlin #501: G29 J command results in 2nd probing point X-coordinates outside of bed limits

Fury1SOG, 2021-04-15:
Hello,

When a G29 J command is performed with E3V2-UBL-BLTouch-15x15-HS-v4.2.2.bin I have noticed that the 2nd probing point coordinates are set to X: 245 Y: 28.1. The X coordinates are outside of the stops at 230mm, but is this intended?

I have a cable chain mount (https://www.thingiverse.com/thing:4316238) installed on the X-carriage which strikes the Z-frame during G29 J.

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

aryajuanda, 2021-04-15:
my X can go to 240 without hit anything.. 
Bed size 230 , Max x pos = bed size + 15.

with X in 245 it can reach far right on the bed (around 200 mm for the probe)

Fury1SOG, 2021-04-15:
In the current configuration, using a direct drive with the extruder motor mounted to the X carriage would have disastrous consequences, but I guess people who use DD compile their own firmware?

Jyers, 2021-04-15:
That is the general assumption yes. The precompiled configs are made to work well with a relatively stock printer, because if you're adding things like direct drive I have full faith you'll be able to handle compiling the firmware to match. Trust me, its worth it for the added customization alone and its not nearly as hard as it is daunting.

closed 2021-04-15 by Jyers