Jyers/Marlin #503: Z-Offest homing gouged my print bed.

nuker1110, 2021-04-15:
### Description

After upgrading the springs on my Ender 3 v2, had to change the Z-Offset to NOT damage my bed due to the height difference. Homing Z0 worked alright, pressing down on the corner of the bed and compressing the spring to reach the Z-stop. Then, to my abject horror, it moved laterally _with the nozzle contacting the bed_ before moving Z to the offset height. Now I have a skid scratched from X0Y0 to the center of my print plate.

### Steps to Reproduce

*Described above

**Expected behavior:** Raise print head to defined Z height before lateral movement on either X or Y axis.

**Actual behavior:** X/Y movement engaged before Z movement, damaging print bed and nozzle.

smooreace, 2021-04-15:
Thats what happens when you dont set your Z stop properly and/or verify
things before you start an automated process.

nuker1110, 2021-04-16:
The question remains, if you're going to move the head to the center of the plate, why find the Z origin at the corner?

And then, why move XY before Z when trying to set the Z offset?

And there was no way for me to know that it was going to move XY first prior to setting the Z offset, which was the entire process I was trying to complete.

Jyers, 2021-04-16:
Good news is, if you're using a brass nozzle like most, your bed is probably fine. Brass is quite soft and so what you perceive as a scratch is most likely just the brass having rubbed off on your bed. 
The issue in this instance was that you should generally level your bed before setting your offset. In that case you would have immediately found all the corners to have been too high and set them lower. That being said I'll add a z lift before the move on non BL Touch configs to help prevent this from happening to others.

closed 2021-04-16 by Jyers

markcarroll, 2021-07-08:
I would like to re-open this as I think it is an issue.  This just scratched my bed too.   

I was starting a manual bed level, selected bottom left and found that the nozzle went down to -0.4 Z which embedded it in the glass.  First thought was that the z-offset was off so I clicked on the Live Adjustment option on the Z-Offset menu which prompted the printer to home the head **without raising the z axis** promptly causing a nice long scrap in my glass bed. 

This is clearly a bug that needs an albeit simple fix.   Just raise Z until it is centered to prevent damage that may occur due to the fact the z-offset or bed is not level, or the user would not be executing that command.

Jyers, 2021-07-08:
I'll look through the config files to see if marlin has anything that forces the z to raise before homing, but if there isn't a mechanism already in place for this there isn't much I can do. If that's the case someone should bring up this issue on the main marlin repo to get it sorted out.

smooreace, 2021-07-08:
Sounds more like improper use. You cant adjust Z offset properly if the
nozzle/probe isn't in a proper location. Since bed leveling is based on the
height of the center of the glass... it travelers there to adjust Z offset.
It did exactly what you told it to do. Should have moved Z to a safe height
and then the homed. Also, if -.04 offset caused a nozzle impact then the
BLT isn't installed properly. It's "supposed to have a 3mmish offset from
the nozzle. And your gouge is much more likely to be brass rubbed off on
the print surface, as it's far harder than the brass is.

This is like saying that a car manufacturer is at fault for an accident
because pressing the brake should inherently force the throttle to zero
automatically to prevent an accident.

markcarroll, 2021-07-08:
I disagree with both that "improper use" assessment and the wholly flawed analogy. 

Clicking on the `Live Adjustment` menu causes the nozzle to move to the center of the bed, fair enough.  However, moving the nozzle across the bed without first lifting it up is very definitely an error.  It was not asked to move due to a simple x/y movement command, it moved due to clicking on the z-offset live adjust checkbox. Regardless of what the user was doing before, that should not happen without a z lift. 

In this case I had switched to Manual Bed Leveling and selected Bottom Left. It moved the nozzle to the right location and then lowered it to the bed. However the Z value read **-0.4**.  That in itself seems like a bug. Why would it not be lowering it to zero? This is manual bed leveling here, not probe-based.  So then I thought, huh, my z-offset must need adjusting and so choose Live Adjust. That's when the movement issue occurred and the nozzle moved to the center while still at -0.4z. 

That shouldn't happen for both reasons outlined above. Nothing to do with driver error.  Glad to hear my scratch is probably just brass, sounds like a good reason to avoid the steel nozzles I keep seeing advertised. 😄

Jyers, 2021-07-08:
> Glad to hear my scratch is probably just brass, sounds like a good reason to avoid the steel nozzles I keep seeing advertised

That's what I say anytime anyone mentions steel nozzles 😅

markcarroll, 2021-07-08:
No kidding. I had no idea of the damage that would do until today.