Jyers/Marlin #510: M112 with 1.3.0 BLTouch-5x5-v4.2.2 when the SD card is in slot

billygr, 2021-04-16:
Hi,

I was using successfully the 1.3.0 E3V2-ManualMesh-5x5-v4.2.2 Ender 3 V2 with no issues at all.

Upgraded to 1.3.0 E3V2-BLTouch-5x5-v4.2.2 (BLtouch works as expected, including mesh etc)

The following behavior is reproducible:
With the SD card in slot
Power up the printer
Connect with Octoprint(OctoPrint 1.5.3 Python 3.7.3 OctoPi 0.18.0)
M112

Shutdown the printer and remove the SDCARD
Power up
Connect
No issues

I haven't encountered any other issues with the BLTouch version during printing, only during boot with the SD in slot

Thank you !!!

ussdeveloper, 2021-04-16:
Experiencing the same problem. Working fine octopi and without sd.

Jyers, 2021-04-16:
Interesting. Good to know that the sd card is part of what's contributing to this bug. This originates in main marlin, so all we can really do is wait for the fix. On the bright side there is a workaround here to prevent the M112 command from being sent https://github.com/Jyers/Marlin/wiki/OctoPrint-Settings

closed 2021-04-16 by Jyers

billygr, 2021-04-16:
Thanks , will search also for it upstream