Jyers/Marlin #517: SD Card Sorting

happycncer, 2021-04-17:
Sd Card is sorted 'alphabetically' rather than 'Most recent First' 

It appears, having checked the source files, that both SDCARD_RATHERRECENTFIRST and  SDCARD_SORT_ALPHA have both been defined as active in the Configuration_adv.h file. It should surely be one or the other defined - not both? 

As it is at the moment, all my files are shown haphazardly on the Ender display - alphabetically rather than by most recent first.

Is it possible to create a binary which adresses this anomaly - I'm sure I can't be alone in finding this an annoying issue?

I am currently running E3V2-BLTouch-3x3-HS-v4.2.2.bin which works superbly - but I do so wish to get the SD Card working correctly.

Jyers, 2021-04-17:
I believe the files are stored alphabetically on the stock creality firmware as well, so it will most likely stay that way on this firmware. I will however look into adding this as an optional setting in the advanced menu.

closed 2021-04-17 by Jyers

Virusmater, 2021-05-20:
Hello @Jyers,
Actually on the stock firmware it sorts by recent first

Jyers, 2021-05-22:
Based on the marlin configuration for the stock firmware, it sorts alphabetically

happycncer, 2021-05-23:
Personally, I'm not really bothered which way is  'supposedly' correct, I would like the choice to be made available to me and would really appreciate this being made as an optional setting in the advanced menu as indicated by Jyers on 17th April

As of this moment, I am numbering all my files to try to make the Most Recent First option work for me - having said that, it is not very good once one exceeds the number 10! Because of that I move the existing files 1-10 into a dated 'Done' folder and start all over again. Messy I agree - but at least it works the way I want it to!

Virusmater, 2021-05-23:
It might be sorting alphabetically by default on the Marlin firmware, but my Ender 3 v2 came with the firmware that had sorting by a date.

Jyers, 2021-05-23:
From what I could tell, it didn't seem like I would be able to expose the sorting as a firmware setting, but I will look into it

happycncer, 2021-05-24:
From info I have read elsewhere (forgotten where from now!) it could be tied in with the fact that SDCARD_RATHERRECENTFIRST and SDCARD_SORT_ALPHA have both been defined as active in the Configuration_adv.h file - and with both being defined as active at the same time, it seems that the SDCARD_SORT_ALPHA takes precedence?

Only needs one or other to be defined - not both?

lpaln, 2021-06-21:
I too am coming to this thread which was closed, but not resolved.

Firstly let me say thank you very much for running this project, and please do not take my comments as critical. This is a seriously useful contribution to the community and I appreciate your efforts.

The OP however is entirely correct. The stock firmware sorts by modified date so that the most recently created .gcode is at the top. In fact, I have many printers and this is always what is shown.

Please consider configuring the auto builds this way in the future as it is the norm that most people have come to expect and makes the most sense.

Jyers, 2021-06-21:
I will put it on my todo list and take a look into it. I had had reports at one point that sorting recent first lagged the menu somewhat, but I'll see if any of that holds water.

lpaln, 2021-06-21:
Thank you. I just installed the firmware this morning and it is the first thing I noticed. I'll report back if I have any other findings and I would be glad to contribute in some way to this project.

I won't be able to for several days as The V2 is going to be occupied for a few days, however I have browsed through the build docs and will gladly install the required plugins to vscode, test the settings mentioned above and report back here if that will keep you free to work on more challenging/impactful things.

Jyers, 2021-06-21:
I'll give it a try when I have time. We'll just see who gets to it first 🙂