Jyers/Marlin #524: Missing X&Y nozzle offset

sbsoneji, 2021-04-18:
Ender 3 v2 nozzle comes with -3mm X and 1-1.5mm Y offset nozzle so if i use it in cura then it sets positions greater than 230mm for prints so prints doesn't come as designed dimensions. Another way is that we substract the offset from max X & Y value, it works fine but we lose printable bed area. So set X&Y offset function is needed in marlin firmware

Tuncay-Ayhan, 2021-04-18:
What exact firmware version are you using?

sbsoneji, 2021-04-18:
> > Ender 3 v2 comes with -3mm X and 1-1.5mm Y offset nozzle so if i use it in cura then it sets positions greater than 230mm for prints so prints doesn't come as designed dimensions. Another way is that in addition to above method we substract the offset from max X & Y value, it works fine but we lose printable bed area. So set X&Y offset function is needed in marlin firmware
> 
> What exact firmware version are you using?

V1.3.0 on Creality V4.2.2 board

Tuncay-Ayhan, 2021-04-18:
> V1.3.0 on Creality V4.2.2 board

1. Do you use a BL Touch?
2. nd what version in 1.3.0 are you using?

sbsoneji, 2021-04-18:
> 1. Do you use a BL Touch?
> 2. nd what version in 1.3.0 are you using?
I am using E3V2-ManualMesh-5x5-v4.2.2.bin firmware

Tuncay-Ayhan, 2021-04-18:
> I am using E3V2-ManualMesh-5x5-v4.2.2.bin firmware

No BL Touch?

sbsoneji, 2021-04-18:
> > > > > > Ender 3 v2 comes with -3mm X and 1-1.5mm Y offset nozzle so if i use it in cura then it sets positions greater than 230mm for prints so prints doesn't come as designed dimensions. Another way is that in addition to above method we substract the offset from max X & Y value, it works fine but we lose printable bed area. So set X&Y offset function is needed in marlin firmware
> > 
> > 
> > > > >
> > 
> > 
> > > > >
> > 
> > 
> > > > > What exact firmware version are you using?
> > 
> > 
> > > >
> > 
> > 
> > > >
> > 
> > 
> > > > V1.3.0 on Creality V4.2.2 board
> > 
> > 
> > > 
> > 
> > 
> > > 1. Do you use a BL Touch?
> > 
> > 
> > > 1. nd what version in 1.3.0 are you using?
> > 
> > 
> > I am using E3V2-ManualMesh-5x5-v4.2.2.bin firmware
> 
> No BL Touch?

BL Touch NOT installed

Tuncay-Ayhan, 2021-04-18:
> BL Touch NOT installed

You should not have to change any settings about the offsets within the firmware. Maybe @Jyers understands the issue?

sbsoneji, 2021-04-18:
> You should not have to change any settings about the offsets within the firmware. Maybe @Jyers understands the issue?

closed 2021-04-18 by sbsoneji