Jyers/Marlin #529: Missing Set Home Offsets Menu Function On LCD Display

sbsoneji, 2021-04-18:
@Jyers 
Stock firmware has "set home offsets function" but V1.3.0 build lacks it.
This version lacks it (in stock firmware it is under prepare section).
This function is very handy as Ender 3 V2 does comes with offset nozzle from bed in X&Y both.
I am currently using E3V2-ManualMesh-5x5-v4.2.2.bin (from V1.3.0)
I have attached images for referance.
[image]
[image]

Jyers, 2021-04-18:
Ahhh, yes. I remember that replaces the Z Offset item in the stock firmware when a probe is not present. I'll most likely add this option to the advanced or motion menu.

Jyers, 2021-04-19:
I have added a Home Offsets menu to the Motion menu. If you compile your own you can get it now, or it will be out in the next release.

closed 2021-04-19 by Jyers