Jyers/Marlin #530: Print Finish Progress is not 100%

xantrk, 2021-04-18:
[image]

Hello, I saw d35b6093c154d65fda7db7d86c4f9d022ae4690d should fix this issue but, I had this screen upon my print finish. The progress is not set to 100% and remaining time increased. I am using Emit M73 option in PrusaSlicer to show accurate remaining time, not sure if it is related. Wanted to report. 

Thanks

Jyers, 2021-04-18:
Just a sanity check, you do compile you're own version right? As this commit was after the last release and therefore is not included in 1.3.0.

xantrk, 2021-04-18:
> Just a sanity check, you do compile you're own version right? As this commit was after the last release and therefore is not included in 1.3.0.

Yes, compiled yesterday. You can see that text has yellow color, which belongs to a later commit

Jyers, 2021-04-18:
I think I fixed this now.

closed 2021-04-18 by Jyers

xantrk, 2021-04-20:
Dear @Jyers , just updated to latest commit and this is what I got. It is completely not a big deal, but just wanted to let you know.

[image]

I believe these guys ( https://github.com/Jyers/Marlin/discussions/540 ) are also have this issue.

vincechiu, 2021-04-20:
Yes, same here, I got 99% with last completed print.

Jyers, 2021-04-20:
Think I found the issue, you all are printing from SD right?

xantrk, 2021-04-21:
> Think I found the issue, you all are printing from SD right?

Yes, I am. Thanks!