Jyers/Marlin #531: Probe releasing too fast for second measurement

f-zappa, 2021-04-18:
Installed this firmware yesterday and already love it, but I found one problem with my BLtouch (clone). When dual-probing, the probe is released when the head is still very close to the bed, making the probe hitting the bed. This might destroy the sensor after some time. Behaviour was different with my old, self-compiled firmware, I compared the configs but I couldn't figure out which setting is responsible for this effect.
I made a slow-motion-video showing the problem.

f-zappa, 2021-04-18:
https://user-images.githubusercontent.com/43571251/115149957-a0370200-a066-11eb-887c-9ddc41111d01.mov

aryajuanda, 2021-04-18:
[image]

you can set the clearance here...

Jyers, 2021-04-19:
Let me know if @aryajuanda solution works. If not I can assist in looking for another potential fix.

f-zappa, 2021-04-19:
I changed this values to 15/10/10, but the behaviour is still the same.

```
#define Z_CLEARANCE_DEPLOY_PROBE    15 // Z Clearance for Deploy/Stow
#define Z_CLEARANCE_BETWEEN_PROBES  10 // Z Clearance between probe points
#define Z_CLEARANCE_MULTI_PROBE     10 // Z Clearance between multiple probes
```
Any other ideas?

aryajuanda, 2021-04-20:
mmm i have no idea.. i m not using normal mode, i use HS (High Speed) mode so my probe nozzle did not push in n out each time..

foxylion, 2021-04-23:
I'm having similar issues with the `E3V2-UBL-BLTouch-10x10-v4.2.2.bin` variant of the latest release.

When letting the printer create the mesh some probes will fail because the second measurement fails due to the probe going into ALARM mode.

So what happens in detail is:

1.  probe deploys for the first time
2.  probe triggers after z was decreased enough
3.  probe is deployed for the second time
4.  probe directly hits the build plate, because z was not already increased enough
5.  probe directly tries a second time to deploy (only a few ms in between), this again fails
6.  probe "retracts" and goes into alarm state
7.  build plate is lowered until it hits the Z stop
8.  a value that is completely off is stored in the mesh

_Before I moved the z stop again into place as an emergency stop the print head hit the build plate twice (due to this issue)._

Is there any option to delay the second deploy for a few more milliseconds? I think that would fix the problem.

I would also recompile the firmware, but I'm currently failing with that (see https://github.com/Jyers/Marlin/discussions/558).
Is it possible to get the exact same configuration files you (@Jyers) used for building `E3V2-UBL-BLTouch-10x10-v4.2.2.bin`?

foxylion, 2021-04-23:
Okay it looks like I found the option that solves the problem for me:
In `Configuration_adv.h` uncomment `#define BLTOUCH_DELAY 500`.

This delays the re-deploy of the BL Touch for 500 ms after every trigger.
This gives the print head enough time to move enough upwards.

Jyers, 2021-04-23:
@foxylion Thanks for finding that! This seams to only be an issue with BL Touch clones, but I'll looking into making that the default. I just need to check if it impacts high speed mode strangely as in that mode it doesn't retract between probes.

closed 2021-04-23 by Jyers