Jyers/Marlin #538: Extruder stops following filament runout

tommymonk, 2021-04-19:
I had a weird issue today following a filament runout.

1) The filament ran out and the runout sensor paused the print and ejected the remaining filament
2) I inserted a fresh filament and selected OK.
3) It fed back in, purged and resumed printing, I confirmed that it resumed at the right position and the filament was getting laid down properly
4) A few seconds later the extruder stops turning whilst the print continued

Here is a video: https://youtu.be/DiChTwnBDC8

I was printing from Octoprint, I'm not sure if it's related but when it resumed printing the screen showed message saying that the print was paused and had an "OK" button even though it was printing.

I'm not sure if this is a Marlin issue or Octoprint

Jyers, 2021-04-19:
My best guess is that this would be a Marlin issue. Let me know if your able to recreate the issue.

tommymonk, 2021-04-19:
Sure, I was just posting in case someone pointed out something I missed, if it happens again I'll create issue upstream, feel free to close

closed 2021-04-19 by Jyers