Jyers/Marlin #548: Glitch in Live movement option

AngleV, 2021-04-20:
### Description
There is a small glitch in the Motion Menu when you select the Live movement option.

### Steps to Reproduce

1. Select  Live Movement.
2. Now try to select to move the extruder.
3. Popup appears informing you to go to preheat menu.
4. Select Cancel.

The redraw back to menu glitched and a new checkbox appears where it shouldn't.

tititopher68, 2021-04-20:
@AngleV Just tested like you but no issue !

AngleV, 2021-04-21:
I'm sorry i gave wrong description.
The Extruder 0.00 value gets to the below "row" next to probe checkbox and dissappeared from Extruder selection. Check the attached photo.

[image]

tititopher68, 2021-04-23:
Very curious indeed! I do not have this issue.

It is true that I am on my own build and my fixes and various additions may not have the same effects as Jyers.
I added a large number of modifications, among other things, for the appearance, partly taking up your idea. I can now have for "Cancel", "Confirm_E" and "Continue_E" all the colors defined for the background or the text (I went through your exchanges on the subject).
I am also finalizing (under test) the Assistant Tramming function (equivalent to Marlin's G35).

I think it will be available on my branch tomorrow or the day after tomorrow.

AngleV, 2021-04-24:
@tititopher68 

Go the cursor all the way down to Live Movement option. Then go up and select Extruder. Click to change the value. Select Cancel on the popup and you will see it.
It does not do it if you dont scroll all the way down to the last menu option.
Some one of the variables has wrong value when the menu has been scrolled but i have hard time to follow the code to figure it out.

tome9111991, 2021-04-24:
i have same issue with default theme

tititopher68, 2021-04-24:
@AngleV I'm sorry, but at home, I don't have this finding ... Everything seems normal ... but I don't have the same build ... If I had this problem, I should have a "0" appear to the right of Live's checkbox ...

To note. I noticed that sometimes the encoder button had a problem, ie it does not react immediately when turning the button to the right or to the left. By restarting the printer and doing the same thing again, the operation of the rotary knob is normal again .... maybe the cause of your problem and the misalignment of the values ...

AngleV, 2021-04-24:
> @AngleV I'm sorry, but at home, I don't have this finding ... Everything seems normal ... but I don't have the same build ... If I had this problem, I should have a "0" appear to the right of Live's [...]

I noticed that too but i am sure that has nothing to do with the current issue.

tititopher68, 2021-04-24:
https://user-images.githubusercontent.com/18162675/115970929-f19f2f80-a545-11eb-9aa4-f51abf1a15c3.mp4

AngleV, 2021-04-24:
@tititopher68 

ok, now i figured this out. 
This is what you are missing here.

My setup has an extra menu item. The Probe item above the Live Movement.
In your setup i dont see it.

By this extra menu the screen has to scroll up to see the last item.

Thats why you dont reproduce the issue.

I am sure @Jyers will understand now and he will fix it soon.

Jyers, 2021-04-24:
@AngleV Thanks for the extra info. I'll take a look and see what's causing it to miss that item

tititopher68, 2021-04-25:
@Jyers : for menu move, case Move_E , change to do : Redraw(true);

Jyers, 2021-04-25:
@tititopher68 You shouldn't need a true there because the heating popup is just a wait, selection is not modified. The fix was in the popup control for the Etemp popup

AngleV, 2021-04-26:
@Jyers 

Glitch still exists 

https://user-images.githubusercontent.com/7430985/116047267-9459ca00-a67c-11eb-9473-579f98797f65.mp4

tititopher68, 2021-04-26:
The fix  : 
case MOVE_E:
            if (draw) {
              Draw_Menu_Item(row, ICON_Extruder, (char*)"Extruder");
              current_position.e = 0;
              sync_plan_position();
              Draw_Float(current_position.e, **row**);

Jyers, 2021-04-26:
There is is, thanks @tititopher68!

AngleV, 2021-04-26:
@Jyers 
@tititopher68 

yep!!!!

closed 2021-04-26 by AngleV