Jyers/Marlin #556: Z axis motor is disabled after the print

Majestry, 2021-04-21:
### Description

### Steps to Reproduce

1. Print from SD card is completed, the confirm was not clicked using the display.
2. The last GCode lines are next (generated with Cura 4.8.0 or 4.9.0)
```
G1 X0 Y230 ;Present print
M106 S0 ;Turn-off fan
M104 S0 ;Turn-off hotend
M140 S0 ;Turn-off bed
M84 X Y E ;Disable all steppers but Z
M82 ;absolute extrusion mode
M104 S0
```

3. The Z axis motor is disabled, and this cause the X gantry to fall and hit the bed with nozzle. However, according to the M84, it should be powered, so X gauntry stays on the position.

P.S. Not sure if this is the marlin general issue or related to your modifications.

tititopher68, 2021-04-21:
Neither one nor the other, nor the firmware but probably your hardware....

Majestry, 2021-04-21:
Not sure if this the hardware issue.
Board 4.2.7, all other stuff is stock except the BLTouch.
Previously i had tighten the wheels, so gauntry stay in place even if the Z motor was disabled.
After last maintenance, i adjusted the tightness of wheels and cleaned/lubricated z rod, it moves much more better and even fall under its weigth. That's when i noticed this issue. 
Will try to record some video later

aryajuanda, 2021-04-22:
it should not fall under it weight even we disable the motor.. its a hardware issue..

Majestry, 2021-04-22:
Again. The root of problem is not in falling gauntry. Even with tight wheels and not falling gauntry - why the Z motor is disabled when print is finished, if no command to disable it was sent?

aryajuanda, 2021-04-22:
sorry i dont know what the goal with that kind of setup.. did it work in creality official firmware? never test to keep the motor on after printing done.. 

i always disable all motor after printing, from slicer also from octoprint (G-Code Script), to make sure the printer is cooling down and ready for next print, now my printer is almost up for 24 h/day..

Majestry, 2021-04-22:
Ok, i got much more detailed steps. 
1. Turn on the printer.
2. Perform any action to make steppers on (e.g., auto home or load filament)
3. Wait till steppers will be disabled by timeout (about 2 minutes).
4. Make steppers active again (e.g., using G0 command from serial or moving any axis from screen)
5. Send M84 X Y E command to serial
6. All steppers will be immediatly disabled.

Jyers, 2021-04-22:
This is actually a hardware limitation I believe. On the Creality boards the xyz steppers drivers all share the same enable pin, so I don't think it's possible to disable just one.

closed 2021-04-22 by Jyers

Majestry, 2021-04-22:
Cool. Thank you for the explanation, and sorry for disturbing.

msdossajo, 2021-04-22:
Install a anti backlash in the lead screw, that will prevent the x 
[image]
carriage to fall when the z stepper motor is off

Jyers, 2021-04-22:
^^^ I use one of these and it provides the perfect amount of drag while also reducing backlash issues.