Jyers/Marlin #557: Z-Offset tool bug when USE_PROBE_FOR_Z_HOMING is undefined

Schnabulation, 2021-04-21:
### Description

(This is a follow up to this discussion: https://github.com/Jyers/Marlin/discussions/505)

This issue started to appear in build 1.3.0 and upwards: When using the Z-offset menu together with "Live Adjustment" the Z-offset seems not to stick and reset after homing. Back on 1.2.1 the issue did not appear.

I did a lot of testing today and found that it is tied to the way my printer is configured:
I am using a BL Touch for auto bed leveling, **however** I am still using the stock Z-axis switch to home my Z-axis.
This can be done by keeping the Z-axis switch attached and commenting out (= disabling) `USE_PROBE_FOR_Z_HOMING`.

Further testing and some going back and forth from 1.2.1 to 1.3.1 I noticed that in the newer builds the Z-offset menu disables mesh leveling, rendering my Z-offset useless. This can be confirmed with the command `M420` in a terminal:
V 1.2.1: Using the Z-offset menu bed leveling stays on all the time
V 1.3.1: Using either "Live Adjustment" or "Home Z-axis" disables mesh leveling.

Also: I did a further test on build 1.3.1 **with** `USE_PROBE_FOR_Z_HOMING` enabled and of course: there the Z-axis tool works flawlessly.

I suspect that because I am using the Z-axis switch for homing my Z home position and my BL Touch trigger position is different and therefore I need the Z-offset menu to keep mesh leveling active.

### Steps to Reproduce

(You need to have a physical Z-axis switch attached and the BL Touch installed on the special BL Touch port.)

1. Compile the firmware with `USE_PROBE_FOR_Z_HOMING` commented out (= disabled)
2. Home the printer
3. Do a auto leveling
4. Try to use the Z-offset menu the find the correct Z-offset
5. Do another auto leveling
6. Go back into the Z-offset menu and find that the Z-offset is wrong again.

### To conclude

I believe a condition to check wether `USE_PROBE_FOR_Z_HOMING` is active or not and in turn keep mesh leveling active or disable it in the Z-offset tool would solve this problem.

#### Additional Information

If needed, find my configuration here: https://github.com/Schnabulation/Marlin

Schnabulation, 2021-04-21:
Further testing revealed that this is exactly the problem! I edited [creality_dwin.cpp](https://github.com/Jyers/Marlin/blob/Ender_3_V2_Extensible_UI/Marlin/src/lcd/extui/lib/creality_dwin/creality_dwin.cpp) and commented line 1047 out, in turn keeping mesh bed leveling on during the usage of the Z-offset tool.

It rendered the tool useful again and behaving the same as in version 1.3.1.

tititopher68, 2021-04-21:
No, this line must be enabled....but didn't you forget to do this : 
#define RESTORE_LEVELING_AFTER_G28
//#define ENABLE_LEVELING_AFTER_G28

Schnabulation, 2021-04-21:
Thank you but yes, I have already enabled that before.

tititopher68, 2021-04-22:
It is written in Configuration.h this:
 `If` using a Probe for Z Homing, enable Z_SAFE_HOMING also!`
 So why would you want to uncomment #define Z_SAFE_HOMING ???

Schnabulation, 2021-04-22:
Please re-read my issue above. It is totally related to something else.

But thanks for your suggestion.

tititopher68, 2021-04-22:
Oooh sorry !!!

tititopher68, 2021-04-22:
If you use the dedicated port (the 5 pins connected) + the Z-endstop, there is interference and malfunction.  So try this unusual connection by reconnecting the 2 wires to the right of the connector to the BLTouch port (normally the black and white wire).  Which would correspond better with your configuration file .... I consider that the most reliable and functional connection is the connection of the BLTouch to the Z-endstop.  There is also an issue with the dedicated port which can be fixed by soldering a resistor of 10KOhms.

Schnabulation, 2021-04-22:
I appreciate your input, I really do - but I do know of the limitations of my setup and also the issue regarding using the specific BL Touch port on the Creality 4.2.2 mainboards.

The problem I mentioned above is clearly software related as I have already found the issue and fixed it in my fork of Jyers firmware. It is now up to him if he wants to support my special use case whereas the stock Z-axis switch keeps getting used for homing (which btw I am a big fan of, because I'm not relying on the BL Touch to home).

Jyers, 2021-04-22:
This is a very interesting issue. I would be interested if this also applied to manual mesh configs as well. The only issue with this change is the fact that the way movement is handled in the menu would cause the mesh offset to be applied every time you set a new value. You can observe this by clicking into modify the z offset value and then clicking again without changing it, the z axis will move even though no modification was made. I'll look into the best way to address this problem.

tititopher68, 2021-04-23:
I would like to make sure of this: when an Autohome is performed or when you activate the Live adjustment (which can also engage an Autohome), the Z-Offset seems to be incorrect?  Without modifying the value of your Z-OFFSET, if you only make a Home Z in the menu again, is the offset still present?
 I never paid attention to this because I always remade a Home Z before setting the Z-Offset.
 Indeed there is an issue after the Autohome.
 The cause in the code seems to be, not the designated line, but: gcode.home_all_axis (true) ...
 When you launch a print, you send through the StartGcode a G28 which to do the Autohome, temporarily deactivates the bed leveling and once finished reactivates it.
 The command introduced by @Jyers since v1.2.2 does the Autohome but keeping the leveling active.  So it would be for me, a gcode.home_all_axis (false) to do ...
 I will do some testing with this modified command.

thefabnab, 2021-05-04:
I may be seeing something related. I'm trying to set my home offset using the v1.3.1 release, I'm finding that the home offsets I'm setting aren't working when I try to auto home the print head.

orangoo, 2021-05-04:
Is this issue the same as the z-offset value used in the Tune menu during a print not being saved?

To clarify what I mean is:
1) set the z offset using in the Tune menu option during a print
2) on completion of print - power cycle the printer
3) turn on printer and notice that the z offset value is set to zero.

Is this what you are describing?

thefabnab, 2021-05-04:
I just installed a microswiss direct drive and when I set the Home Offsets on the printer they don't seem to be going to the correct position when I auto home... but now that I'm thinking about it G28 should home all the axis, not place the head where the 0,0,0 position is taking into account the offset.

I'd say either ignore or delete my post. It isn't relevant to this issue and likely isn't an issue at all.

tititopher68, 2021-05-04:
@thefabnab, It is not the same subject here ....
It's not an issue either ... just look at this video: https://youtu.be/8MQRxBISYmU

Jyers, 2021-05-08:
@Schnabulation Sorry it's taken me so long to get back to this. Just a question, when you have commented line 1047 out, can you test going into the z offset menu, and with live adjust on, try just clicking in and out of the value modification without changing it. One of the primary reasons I disabled leveling in this menu was to prevent a strange inconsistency that came from the mesh offset being applied every time you set the position.

Schnabulation, 2021-05-10:
@Jyers No worries, Jacob! With my modification I can run the firmware perfectly fine and have a reliable Z-offset.

I have done as you asked: I have not noticed any change in the Z-offset without modification of the value. Here is how I tested:

1. Auto-home and auto-level. Save mesh to EEPROM.
2. Went into Z-offset menu, activated "Live Adjust" --> Printhead moved to center, auto-homed Z.
3. Mesured the offset with a piece of paper (paper barely scratching).
4. Clicked into the Z-offset, confirmed value **without changing it** --> paper barely scratching, no change
5. repeat step 4 a couple of times
6. Z-offset did not change at all --> paper still barely scratching.

tititopher68, 2021-05-10:
@Schnabulation : It's really is a very strange thing.  I did your procedure, but without making your change in the firmware.  Z-Offset did not change at all ...
 Precision, I always heat my hotend before.  With a cold hotend, i have a 0.05mm gap.  This is normal, the brass nozzles always dilate a little.

Schnabulation, 2021-05-10:
Did you use your BL Touch to home the Z-axis or are you using the Z-axis switch?

tititopher68, 2021-05-10:
@Schnabulation The BLTouch is connected via the Z-endstop.  I think your problem using the firmware without your mod, is the difference in accuracy between the BLTouch and the Switch.  You use 2 different elements to take the 0 point and the Z-Offset and the less precise element takes over.  You must therefore then keep the correction of the Mesh to compensate for the difference in precision and therefore you must actually do your motivation to obtain this.  Your connection method is very personal and atypical, Marlin did not necessarily imagine such a use.

Schnabulation, 2021-05-10:
Thing is: I have found a solution and am using this already successfully with my personal firmware. If Jacob does not want to implement this, I'm fine with it.

zainag, 2021-07-29:
> 
> 
> I am not sure if this is part of this conversation, but I tried 5x5 for 4.2.2 & 3x3 for 4.2.2 on my ender 3 Pro, When I do mesh visualizer with Octoprint, the firmware sends out an error and needs the printer and raspberry pi to restart. Sometimes, as the OP mentioned, the Bltouch probe doesn't fire, and the nozzle bites into the bed. Has anyone had this issue?

I have the exact same issue for 5x5 for 4.2.2 - the BlTouch doesn't find z home and the nozzle just crashed into the bed. This is a serious issue that needs to be resolved.

Jyers, 2021-07-29:
If the probe is not triggering properly that is not a firmware issue. Its a hardware issue. There are many threads looking into the problems with Creality boards and bed probes, there are fixes such as the resistor mod or upgrading your board. Hopefully that clears it up for you and you can start looking to diagnose your issue.