Jyers/Marlin #562: Pop-up highlight colour not following menu selection

gothcha, 2021-04-22:
As I was taking the photos @Jyers requested I noticed that the Confirm/Cancel pop-up and the Continue pop-up had the default highlight rather than the colour specified in the Visuals Menu. This makes it difficult to see.

[image]

I edited the following in creality_dwin.cpp to define the correct colours

void CrealityDWINClass::Popup_Select() {
  _const uint16_t c1 = (selection==0) ? GetColor(eeprom_settings.highlight_box, Color_White) : Color_Bg_Window,
                 c2 = (selection==0) ? Color_Bg_Window : GetColor(eeprom_settings.highlight_box, Color_White);_

void CrealityDWINClass::Draw_Print_confirm() {
  Draw_Print_Screen();
  process = Confirm;
  popup = Complete;
  DWIN_Draw_Rectangle(1, Color_Bg_Black, 8, 252, 263, 351);
  DWIN_ICON_Show(ICON, ICON_Confirm_E, 87, 283);
  _DWIN_Draw_Rectangle(0, GetColor(eeprom_settings.highlight_box, Color_White), 86, 282, 187, 321);
  DWIN_Draw_Rectangle(0, GetColor(eeprom_settings.highlight_box, Color_White), 85, 281, 188, 322);_

This is how the pop-ups look now
[image]
[image]
[image]

AngleV, 2021-04-22:
Very nice. We missed that. 

@gothcha  Thank you

gothcha, 2021-04-22:
My pleasure, in fact I was trying to improve the 'Confirm' icon as it is very hard to see the white on the blue, I created some new icons but was unable to incorporate them into the 9.ICO, they come out blank. I tried Gimp and Affinity Photo and I know the .jpg format is tricky.

These are the proposed icons

[image]
[image]
[image]

The fixed highlight box will make it easy to determine the active icon and the black on blue should be more visible.

Maybe you can merge them into the 9.ICO for me and if you like them include in your set?

AngleV, 2021-04-22:
a-haaa.. 

This is nice to hear it because in fact i was thinking the same thing about those icons but i was too bored to fix them.
I will merge them and i will upload a new 9.ico file for everyone interesting in the next days.

Jyers, 2021-04-22:
Thanks for the catch, either me or @AngleV will incorporate the fix into his latest PR.

closed 2021-04-22 by Jyers

tititopher68, 2021-04-23:
I added a large number of modifications, among other things, for the appearance, partly taking up your idea. I can now have for "Cancel", "Confirm_E" and "Continue_E" all the colors defined for the background or the text (I went through your exchanges on the subject).

I think it will be available on my branch tomorrow or the day after tomorrow.

gothcha, 2021-04-23:
@tititopher68 sounds neat - look forward to testing it. Am I correct in thinking you are now not using the 9.ICO icon but instead printing boxes and text? Sounds a good plan and more adaptable, nice work.
This firmware is getting so slick! Just love your ideas.

tititopher68, 2021-04-23:
it's exactly that ! and as I work professionally in the disability sector with visually impaired people, I have also made the modifications to facilitate their visual access as much as possible.

tititopher68, 2021-04-23:
void CrealityDWINClass::Draw_Popup(const char *line1, const char *line2,const char *line3, uint8_t mode, uint8_t icon/*=0*/) {
  if (process == Menu && mode == Popup) last_selection = selection;
  process = mode;
  Clear_Screen();
  const uint16_t color_bg = GetColor(eeprom_settings.popup.bg, Color_Bg_Window);
  DWIN_Draw_Rectangle(0, GetColor(eeprom_settings.popup_highlight, color_bg), 13, 59, 259, 351);
  DWIN_Draw_Rectangle(1, GetColor(eeprom_settings.popup.bg, color_bg), 14, 60, 258, 350);
  uint8_t ypos;
  if (mode == Popup || mode == Confirm)
    ypos = 150;
  else
    ypos = 230;
  if (icon > 0) 
    DWIN_ICON_Show(ICON, icon, 101, 105);
  DWIN_Draw_String(false, true, DWIN_FONT_MENU, GetColor(eeprom_settings.popup_text, Popup_Text_Color), GetColor(eeprom_settings.popup.bg, Color_Bg_Window), (272 - 8 * strlen(line1)) / 2, ypos, F(line1));
  DWIN_Draw_String(false, true, DWIN_FONT_MENU, GetColor(eeprom_settings.popup_text, Popup_Text_Color), GetColor(eeprom_settings.popup.bg, Color_Bg_Window), (272 - 8 * strlen(line2)) / 2, ypos+30, F(line2));
  DWIN_Draw_String(false, true, DWIN_FONT_MENU, GetColor(eeprom_settings.popup_text, Popup_Text_Color), GetColor(eeprom_settings.popup.bg, Color_Bg_Window), (272 - 8 * strlen(line3)) / 2, ypos+60, F(line3));
  if (mode == Popup) {
    //DWIN_ICON_Show(ICON, ICON_Confirm_E, 26, 280);
    //DWIN_ICON_Show(ICON, ICON_Cancel_E, 146, 280);
    DWIN_Draw_Rectangle(1, GetColor(eeprom_settings.ico_confirm_bg, Color_Light_Cyan), 26, MBASE(5) - 6, 125, MBASE(5) + 41);
    DWIN_Draw_String(false, false, DWIN_FONT_HEAD, GetColor(eeprom_settings.ico_confirm_txt, Color_White), GetColor(eeprom_settings.ico_confirm_bg, Color_Light_Cyan), 26 + ((99 - 7 * STAT_CHR_W) / 2), MBASE(5) + 10, F("Confirm"));
    DWIN_Draw_Rectangle(1, GetColor(eeprom_settings.ico_cancel_bg , Color_Light_Blue), 146, MBASE(5) - 6, 245, MBASE(5) + 41);
    DWIN_Draw_String(false, false, DWIN_FONT_HEAD, GetColor(eeprom_settings.ico_cancel_txt, Color_White), GetColor(eeprom_settings.ico_cancel_bg, Color_Light_Blue), 146 + ((99 - 6 * STAT_CHR_W) / 2), MBASE(5) + 10, F("Cancel"));
    selection = 0;
    Popup_Select();
  }
  else if (mode == Confirm) {
    //DWIN_ICON_Show(ICON, ICON_Continue_E, 87, 283);
    DWIN_Draw_Rectangle(1, GetColor(eeprom_settings.ico_continue_bg , Color_Light_Cyan), 87, MBASE(5) - 6, 186, MBASE(5) + 41);
    DWIN_Draw_String(false, false, DWIN_FONT_HEAD, GetColor(eeprom_settings.ico_continue_txt, Color_White), GetColor(eeprom_settings.ico_continue_bg, Color_Light_Cyan), 87 + ((99 - 8 * STAT_CHR_W) / 2), MBASE(5) + 10, F("Continue"));
    DWIN_Draw_Rectangle(0, GetColor(eeprom_settings.popup_highlight, Select_Color), 86, MBASE(5) - 7, 187, MBASE(5) + 42);
    DWIN_Draw_Rectangle(0, GetColor(eeprom_settings.popup_highlight, Select_Color), 85, MBASE(5) - 8, 188, MBASE(5) + 43);
  }
}

void CrealityDWINClass::Popup_Select() {
  const uint16_t c1 = (selection==0) ? GetColor(eeprom_settings.popup_highlight, Select_Color) : GetColor(eeprom_settings.popup.bg, Color_Bg_Window),
                 c2 = (selection==0) ? GetColor(eeprom_settings.popup.bg, Color_Bg_Window) : GetColor(eeprom_settings.highlight_box, Select_Color);
  DWIN_Draw_Rectangle(0, c1, 25, MBASE(5) - 7, 126, MBASE(5) + 42);
  DWIN_Draw_Rectangle(0, c1, 24, MBASE(5) - 8, 127, MBASE(5) + 43);
  DWIN_Draw_Rectangle(0, c2, 145, MBASE(5) - 7, 246, MBASE(5) + 42);
  DWIN_Draw_Rectangle(0, c2, 144, MBASE(5) - 8, 247, MBASE(5) + 43);
}

gothcha, 2021-04-23:
Great well done, coincidentally we have a guide dog puppy which we are training, so do quite a bit with UK Guide Dogs and have quite a few VI friends. 
I will give this code a try and let you know how it looks. I can take some more photos if it would be helpful to it demonstrate the new look. Damian

tititopher68, 2021-04-23:
void CrealityDWINClass::Draw_Print_confirm() {
  Draw_Print_Screen();
  process = Confirm;
  popup = Complete;
  DWIN_Draw_Rectangle(1, Color_Bg_Black, 8, 252, 263, 351);
  //DWIN_ICON_Show(ICON, ICON_Confirm_E, 87, 283);
  DWIN_Draw_Rectangle(1, GetColor(eeprom_settings.ico_confirm_bg , Color_Light_Cyan), 87, MBASE(5) - 6, 186, MBASE(5) + 41);
  DWIN_Draw_String(false, false, DWIN_FONT_HEAD, GetColor(eeprom_settings.ico_confirm_txt, Color_White), GetColor(eeprom_settings.ico_confirm_bg, Color_Light_Cyan), 87 + ((99 - 7 * STAT_CHR_W) / 2), MBASE(5) + 10, F("Confirm"));
  DWIN_Draw_Rectangle(0, GetColor(eeprom_settings.popup_highlight, Select_Color), 86, MBASE(5) - 7, 187, MBASE(5) + 42);
  DWIN_Draw_Rectangle(0, GetColor(eeprom_settings.popup_highlight, Select_Color), 85, MBASE(5) - 8, 188, MBASE(5) + 43);
}

tititopher68, 2021-04-24:
@gothcha 
[image]
[image]
[image]

tititopher68, 2021-04-24:
@gothcha : I had to make some adjustments: decrease the values for Y by 20 points, it's OK now.

AngleV, 2021-04-24:
@tititopher68 

Thats nice !!!

gothcha, 2021-04-24:
Neat, is the code on your repo? When I checked earlier I couldn’t find the relevant .cpp & .h files. I would love to give it a go

gothcha, 2021-04-24:
Can you change the background of the Cancel button too? This code makes it flexible and no need to change icons. The text is very clear.

tititopher68, 2021-04-25:
Also for the Cancel Button. I can change 25 parameters, 20 Colors.
And for me all the issue from AngelV are fixed...
Now i am testing the possibility to change directly in the UI, the mains icons...

tititopher68, 2021-04-25:
I am updating my build tomorrow on the Github

tititopher68, 2021-04-25:
@gothcha I am testing the possibility to change directly in the UI the main's icons and use pack of news main's icons.
But now on my github, the code (without this possibilty in test) is on line ;-)

gothcha, 2021-04-25:
Excellent thank you I shall try and test it out :)

gothcha, 2021-04-27:
@tititopher68 Just compiled your code, it works perfectly! Wow yes there are a lot of choices now. I have configured it to my liking and it certainly solves the original visibility issue. I shall keep this for now until @Jeyers has a new release. There are so many updates/improvements it is hard to keep up!! Thank you

tititopher68, 2021-04-27:
Thank you for your reply !
 I had done what Jyers is proposing today, 4 days ago, but given the simplification of the code proposed today, I have already integrated a part of it.  I have already modified it to correct what I do not find logical and which have an impact on the quality of the first layers  I will make the suggested modifications.
 I will propose a new build offering to visually customize the interface completely.

tititopher68, 2021-04-28:
@gothcha : New build on line

gothcha, 2021-05-04:
@AngleV I have enhanced a few more of the icons and added one for the Probe to use in the advanced menu. I cleaned up the warning icons and the BLTouch icons as they were rather fuzzy and dull!

I successfully packed and loaded these after editing them using Paint.

[image]
[image]
[image]
[Custom Icon data.zip](https://github.com/Jyers/Marlin/files/6420975/Custom.Icon.data.zip)

tititopher68, 2021-05-04:
@gothcha : Very cool !!!!
Just implemented in my build V1.4.4.b (with others things...)

gothcha, 2021-05-04:
Great, nice to have a bit of colour!

AngleV, 2021-05-04:
@gothcha 
it looks awesome !!!

gothcha, 2021-05-04:
Thank you, I will keep working to perfect them and keep a design pattern going. Gradually we will lose some of the blue on grey look! Any suggestions will be taken on board guys.

Jyers, 2021-05-04:
@gothcha Those look great! I'm going to be organizing the display firmware folder soon and making a place for different "themes" or icon packs people have made. Would love to have this be one of them if that's cool with you :)

gothcha, 2021-05-04:
Thanks Jacob, I would be honoured, it is a pleasure to help your great work, it must take up a lot of your precious time! I will work on completing the set and send you a final copy. I know having a distinctive look is important and sets this repo apart. It appears to be exploding based on the increase in the comments & questions.

gothcha, 2021-05-05:
OK guys I have cleaned up and enhanced most of the icons. The one I love in particular is the speed icon. I could never understand the original one, so now there is a cute racing hare!! I quite like the overall look, without going over the top! Let me know if you have any comments or suggestions.

This is the main selection.
[image]
[image]
[image]
[image]
[image]

I will use it for a few days and see if there are any issues then I will release a file set to you Jacob.

Tuncay-Ayhan, 2021-05-05:
> OK guys I have cleaned up and enhanced most of the icons. The one I love in particular is the speed icon. I could never understand the original one, so now there is a cute racing hare!! I quite like [...]

That looks awesome! Thanks for sharing!

Jyers, 2021-05-05:
Looks great! Can't wait :)

tititopher68, 2021-05-05:
What do you think about that ?

https://user-images.githubusercontent.com/18162675/117206913-af9ea500-adf3-11eb-9368-63a597ceab49.mp4

Tuncay-Ayhan, 2021-05-05:
> What do you think about that ?
> 

Damn. That looks amazing. Don't tease me like that xD

gothcha, 2021-05-05:
@tititopher68 Well done!

tititopher68, 2021-05-06:
With the Jyers icon pack (including the new AxisBL icons, etc.) there is a bug.
In principle, the screen firmware only accepts a limited size for the 9.ICO file. And there, if you go to menu Prepare -> Manual Level, the menu is not displayed completely and hangs!

tititopher68, 2021-05-06:
For my part I deleted the 4 Chinese icons entries (# 82, # 84, # 86, # 88) in 9.ICO and in DWIN_ICO.py and more bugs!

gothcha, 2021-05-06:
I agree, it is the same for me, it doesn't hang but the icons are not displayed or mis-formed!! I am sure it was working previously. I shall test with the official DWIN and see what happens

gothcha, 2021-05-06:
I tried re-saving the jpg files and this fixed the one that was distorted but the others are still not displaying. Do you know what the file limit size is?

gothcha, 2021-05-06:
Interestingly when I initially added the Probe Icon as 255 it didn't work! So maybe it is related. The one that works AxisBR is 250. I don't know why there is a gap in the numbers above 205. Maybe try moving them to lower numbers?

gothcha, 2021-05-06:
Christophe - when I used the 9.ICO from the 'Mlevel-arrows' folder it works! This is 230KB the one that I modified that does not is 236KB.
Did you try the one from this folder?

gothcha, 2021-05-06:
Looks like you are correct, if I keep the file to 225KB it is fine. Some of the new icons are larger files due to colour etc. Will have to do some more work to keep them below the limit!

tititopher68, 2021-05-06:
@gothcha :  inf to 227 KB seems the limit

> I agree, it is the same for me, it doesn't hang but the icons are not displayed or mis-formed!! I am sure it was working previously. I shall test with the official DWIN and see what happens

This already explains why @ tome9111991 wrote: "Small Bug: Center Icons looks like not centered i dont know why."

Jyers, 2021-05-06:
We can drop all the text image files to save space. I swapped all image copy text for string draws a while ago.

gothcha, 2021-05-06:
That’s a big help, I was about to try this too, I will go through the .h file and see which icons Jacob is not actually using, I found one yesterday as I was editing them

tititopher68, 2021-05-06:
Yes, it's an idea, but when we create the 9.ICO file with makeIco.py, it add at the ID of the icon  :  UNKOWN. I will also look to modify MakeIco.py to not have this.

gothcha, 2021-05-06:
@Jyers If you use Christopher’s mods to replace the Confirm, Continue and Cancel popups that will save three more!

tititopher68, 2021-05-06:
For my part I deleted the 4 Chinese icons entries (# 82, # 84, # 86, # 88) in 9.ICO....The firmware is only in english

gothcha, 2021-05-06:
I will do the same, I have already made some changes to the .py file

tititopher68, 2021-05-06:
> @Jyers If you use Christopher’s mods to replace the Confirm, Continue and Cancel popups that will save three more!

It's an  idea, not necessarily like for me who works with visually impaired to have color choices (Even if I regret that with most developers, the Accessibility part "is forgotten ....).

tititopher68, 2021-05-06:
The name of the icon is not used in the encoding 9.ICO

tititopher68, 2021-05-06:
But you can create 3.ICO et 6.ICO and 9.ICO with a size max for each ICO 226 KB. The max size is 512 KB for the 3 jpg files  and ICO's files....
Soo i mod the firmware to use  the news icons into 3.ICO

tititopher68, 2021-05-06:
I just make a 3.ICO + 9.ICO
9.ICO is the stock icons
3.ICO is a mix of AngelV, Gotcha and tome911199
I create a function GetPackICO like GetColor and it's OK. I can have now 3 packs...

Jyers, 2021-05-06:
@tititopher68 Interesting, I'll have to take a look at that...

tititopher68, 2021-05-06:
@Jyers 
The differents parts to add for creality_dwin.cpp and creality_dwin.h
[GetPackICO.zip](https://github.com/Jyers/Marlin/files/6436266/GetPackICO.zip)

gothcha, 2021-05-07:
@tititopher68 @Jyers 

[Gotcha_Custom Icons.zip](https://github.com/Jyers/Marlin/files/6440027/Gotcha_Custom.Icons.zip)

I have attached my updated icons. Incorporate these as you think best, it looks like you have a great handle on the DWIN file structures now.

As a note I found that the current icon choices may not be ideal so I made the following changes

Draw_Menu_Item(row, ICON_Speed, (char*)"Max Speed", NULL, true);

case TUNE_FLOW:
     if (draw) {
     Draw_Menu_Item(row, ICON_MaxSpeedE, (char*)"Flow Rate");

case MOTION_FLOW:
    if (draw) {
    Draw_Menu_Item(row, ICON_MaxSpeedE, (char*)"Flow Rate");

I think I am correct that 'Flow Rate' relates to the extruder?

Good luck and I look forward to seeing the final release.

tititopher68, 2021-05-07:
Finally due to the memory capacity, I can only have 2 packs (including the new @Gotcha icons).  So I simplified the code.  In addition I did the firmware migration to Marlin 2.0.8.0 from this day.  I am testing for a few days and will then put it all online.

gothcha, 2021-05-08:
@tititopher68 I was able to use all my icons and the manual level additions in a single 9.ICO by deleting the chinese and other unused icons. I will use this setup for now. Marlin 2.0.8 update list is amazing, I assume almost all of it was in the latest bug fix. Makes me realise why odd things sometimes happen when Jyers releases new versions inc upstream mods!

Jyers, 2021-05-08:
@gothcha What icons did you delete? When I was organizing the icon packs, I noticed even with the Chinese and unknown icons deleted, the .ICO file was still over 226, which @tititopher68 determined to be the max.

gothcha, 2021-05-08:
I deleted 34,77,82,83,84,86 & 88 This saved around 10K. From my checks you were not calling any of these.
 
[9.ICO.zip](https://github.com/Jyers/Marlin/files/6445987/9.ICO.zip)

My addition/updates then took the file to 217K

[8 more replies]